rogerclarkmelbourne/OpenGD77 #10: Better Tx power control UI etc needed

rogerclarkmelbourne, 2019-11-01:
Currently the power is controlled in the Utilities screen, which just adjusts the DAC value, with a limit to prevent the output power on 144Mhz being above approximately 5W.

We need to use the power calibration values, so that the user control the power output, perhaps in 1W steps where power >= 1W and below 1W, perhaps steps of 250mW or even 100mW steps, so that hotspot users can adjust the power.

Also, a better UI is needed for this. Probably Function + Left and Function + Right on Channel or VFO screens to increase and decrease power

alexernst, 2019-11-10:
The power level per channel from the codeplug is not used at the moment.
There should be a mapping like High = 5W and Low = 1W.
Or perhaps the CPS could be extended to support more steps for power level.

rogerclarkmelbourne, 2019-11-10:
That’s true

No one seems to have asked me for this functionality.

It may not be important for Ham radio operators.

I don’t think we can simply use that value to set 1 or 5W , because it looses the flexibility of the multiple power levels.

I will look in the codeplug data, but I think they probably just used one binary bit as a flag for this and we now need 4 bits.

I think there are some unused properties in the Channel struct, which we could potentially use for the power setting, and I could modify the CPS in the same way I modified the CPS for the Squelch levels.

I will post to my blog to ask the question

rogerclarkmelbourne, 2019-11-13:
Fixed

closed 2019-11-13 by rogerclarkmelbourne