rogerclarkmelbourne/OpenGD77 #4: Occasional flutter on signal dropout on DMR Rx

rogerclarkmelbourne, 2019-11-01:
On some radios, the DMR signal seems to drop out approximately every 20 to 30 seconds.

When this happens the audio breaks up, the green led goes out and the display briefly goes back to not displaying the QSO data.

This bug seems specific to individual radios, including the radio that VK4APF has and also F6GVE
On other radios the problem does not seem to occur at all.

My current best guess is that this is an Rx calibration issue, but I don't know what calibration is used during Rx, if any.

rogerclarkmelbourne, 2019-11-05:
This problem seems to sometimes be caused by miss calibration of the radio.

For some reason the OpenGD77 firmware requires more accurate calibration of the radio than the official firmware.

Also. In one instance the radio appears to have a small but  unexplained offset between the Tx and Rx frequencies

darkteckno, 2019-11-06:
I believe I have one of the radios that have this issue. It did have the official firmware updated that was pulled recently by radiodity and a few factory resets. I wonder if this corrupted any calibration data.

rogerclarkmelbourne, 2019-11-06:
Yes.

The latest official firmware can sometimes corrupt the calibration 

Try disabling the calibration in the Options menu.

However this is not a ideal fix.

You probably need to load various other peoples calibration data and see if any of them suit your radio.

lelkluk, 2019-11-15:
Concerning my two faulty GD-77, the last uprade I did with the official firmware was 3.2.2. I never upgraded them with 3.2.4.    If this can help investigating.
Adjusting the TXoffest in Pistar helps a lot, as the DMR drop occurs much less often, but this workaround is not 100% effective.
May be the offset could be better, but as it is not possible to see the BER on the GD77 it is difficult to adjust it more precisely.

rogerclarkmelbourne, 2019-11-15:
There may be a calibration setting which the firmware is not reading from the calibration data in Flash.
Or it could be that the way the OpenGD77 uses the C6000 requires more accurate calibration.

Strangely I have 5 GD-77 radios and don’t see this issue on any of them, and no one locally to me has this problem, otherwise I would gladly exchange radios with them, or probably just adjust the calibration and potentially fix the problem

testbird, 2019-12-07:
This is exactly the type of not-consistently-reproducible, occasional flutter or garbling, that seems to be experienced by some operators, that I'd expect under the hypothesis of possible RF interference and timing artifacts in conjuction with DMR. (Which already came up at https://github.com/rogerclarkmelbourne/OpenGD77/pull/166)

What do you think?

I'll try to separate different aspects that may possibly increase the RX robustness.

**Interference during RX**
* distant repeater/station with direct path and delaying, long reflection path
* local repeater/station (even near field) and distant reflection
* sitting in between reflections with signal traveling back and forth

Problem?:
Without strict timeslot filtering (promiscuous mode) a delayed signal (duplicate, possibly partly mangled frame) may disturb the audio decoding stream.

Remedy?:
RX sequence filtering, picking only frames from a current originator ID in conjunction with the correct frame sequence number.
* Instanciate status data: current ID + sequence ring bitmap + bitmap ring pointer
* RXed frames can set their unset bitmap sequence bit, or are droped otherwise
* Accepted frames unset the oldest sequence bit in the ring, and increase the pointer to allow receiving a next frame

**Interference during TX**

Problem?:
Reflection of the own signal may come back during the other (RX) timeslot.

Remedy?:
Only during TX, drop all packages RX from the TX ID.
(Needs to be separate from above RX sequence filtering, to still be able to filter a incoming full-duplex frame sequence.)

**Interference shortly after TX**
One ore more own data frames may still be coming back from a repeater or reflections.

Remedy?:
A short timeout for muting or inhibiting the reception of frames from the ID that was used in the own transmission.

testbird, 2019-12-07:
> [Remedy for interference during TX] (Needs to be separate from above RX sequence filtering, to still be able to filter a incoming full-duplex frame sequence.)

Actually, as only one timeslot is used for RX, and if only a sigle frame fits into a timeslot it might be possible to re-use the same seqence filtering code. (If there is no way another a delayed frame can possibly be received, just keep the entire sequence bitmap set for the current TX ID during TX.)

testbird, 2019-12-07:
Oops, sorry...
During TX, simply doing the same RX blocking of the current TX ID, as during the short timeout after the TX, should suffice. (If the tolerances are such that only one data frame fits into the single (other) RX timeslot, impeding any duplication.)

testbird, 2019-12-07:
Well, judging from the field experience, interferences seem to only to create flutter occasionally and in certain environments, so robustness measures aren't that generally urgent. But I found one DMR environment configuration that might depend on de-duplication.

That is, operating in conjunction with a single frequency repeater (https://github.com/hamdv/OpenGD77/issues/57).

THAT seems like an ideal test environment for timing and interference robustness measures. And also a nice way to actually determine if deduplication is really needed *in the firmware*, or if the ambe decoder will already handle this.

testbird, 2019-12-07:
> Strangely I have 5 GD-77 radios and don’t see this issue on any of them, and no one locally to me has this problem, otherwise I would gladly exchange radios with them

Strange, or, this might mean that the problem is also associated with the local reflection and signal interference conditions. (Without negating an effect of device calibration.)

rogerclarkmelbourne, 2020-03-02:
This has been fixed by the change to only reset the DMR chip if now DMR signal as been received in the last 390mS (13 frames)

closed 2020-03-02 by rogerclarkmelbourne