rogerclarkmelbourne/OpenGD77 #5: Battery drain on Rx is probably higher than the official firmware

rogerclarkmelbourne, 2019-11-01:
Just a note that the battery seems to go flat faster with this firmware than the official firmware.

Its possible that the Flash and the EEPROM chips are put into low power mode when not in use or the official firmware does some other tricks to reduce the current consumption, but we'd probably need to run the radio though a current meter to check whether there is really any difference.

It looks like the official firmware turns the AT-1846 on and off to save power, because the AT-1846 takes around 30mA according to its spec, which is a significant proportion of the total current used by the whole radio

lasantiago, 2019-11-04:
On my GD-77 with the Nov 4 update, it seems part of the drain has to do with the screen turning on every time the repeater transmits (with the green light turning on but no sound coming out, probably because of the repeater asking for a radio check or beacon). Also,

rogerclarkmelbourne, 2019-11-04:
There are many causes of the increased power usage.

However I think the RF chip constantly receiving takes around 30% more power than in the official battery saver function.

Also I think the C6000 DMR DSP also has a low power mode, which could be used to save power.

I am not sure about the differences between the official firmware and the OpenGD77 for the screen light and LED when a signal is received, perhaps the official firmware has a delay before it turns in the backlight of the display

darkteckno, 2019-11-06:
My gd-77 never shows 100% even after being fully charged with different batteries.

rogerclarkmelbourne, 2019-11-06:
what voltage is your battery when it’s at maximum charge.

It’s likely your charger is not producing 8.4V

darkteckno, 2019-11-07:
The charger is producing 8.51V and the battery at full is 7.5V  58%
according to the gd-77 (7.9v by multimeter)

rogerclarkmelbourne, 2019-11-07:
If you remove the GD-77 from the charger, what voltage is the battery using your multimeter and what voltage does the Batter screen read ?

It should be around 8.3V - 8.4V to be fully charged.

darkteckno, 2019-11-07:
7.5V by gd-77 screen an 7.9V on multimeter, on the back of the battery its
rated at 7.2V

rogerclarkmelbourne, 2019-11-07:
The battery voltage that the firmware shows is the voltage on the CPU, its likely to be lower than the battery voltage on its outside terminal.

If you battery charger won't charge to more than 7.9V it needs to be adjusted so it charges the battery to around 8.4V

darkteckno, 2019-11-07:
Sorry 7.4V on back of battery, I don't understand why we are aiming for
8.4V or are there more than on type of battery?

darkteckno, 2019-11-07:
I tested the battery by its inside terminals. I have another battery
charger that I've not used yet that I will try when home.

rogerclarkmelbourne, 2019-11-07:
Please read about 3.7V LiPo cells and their charging voltage.

darkteckno, 2019-11-07:
OK will do, the other charger I have is a USB charger without wall wart

Wireheadbe, 2020-01-11:
Just chiming in, for those who have issues getting the batteries full. The GD-77 charger, charges the two lithium cells in series. Altough there are 3 terminals at the back, the middle "floating" terminal isn't connected. So you basically can't balance the cells. This is what's causing issues for some people. I've opened all my batteries, and connected this terminal to the "middle" of the two cells. This effectively allows me to balance the cells. I've modified the base of my GD77 with two charge cables, and a 3-pin JST header so it allows connection to a real balance charger. I can get the cells to 100% without hassle, and it's much, much safer. (No overvoltage!)

rogerclarkmelbourne, 2020-01-12:
@Wireheadbe 

Balancing the battery will be better than a non balanced batter, but the main cause of less battery life is that the official firmware shuts down the MCU and all chips except the display and does "micro sleeps" of a few hundred milliseconds, and only wakes briefly to check the RSSI.

We will eventually implement this level of power saving in the OpenGD77 but there are higher priority bug fixes and enhancements to do first.

arekm, 2020-04-03:
> Also I think the C6000 DMR DSP also has a low power mode, which could be used to save power.

Aren't these registers the same as in C5000?

https://www.qsl.net/dl4yhf/RT3/HR_C5000_english.html
https://www.docin.com/p-191686571.html?qq-pf-to=pcqq.c2c ( + google translate app in camera mode)

0x0d register and sleep/suspend there

rogerclarkmelbourne, 2020-04-03:
There is a low power mode, and also a separate signal called POWER DOWN

The transceiver chip also takes a lot of current (30mA according to the spec), and the CPU is probably taking 50mA

But any power saving is very low on the To Do list, as even basic stuff like DMR reception is not 100% the same as in the official firmware, and SMS are not supported, or DCS or VOX or DTMF Rx etc etc

rogerclarkmelbourne, 2020-04-30:
I'm closing all enhancement issues

rogerclarkmelbourne, 2020-10-04:
Re-opened as part of enhancement requests cleanup

rogerclarkmelbourne, 2021-02-06:
[OpenGD77_Experimantal_Eco_Levels_202102061316.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/5936125/OpenGD77_Experimantal_Eco_Levels_202102061316.zip)

This is an experimental version which implements various power saving strategies.

There is a new setting for "Eco Level" on the Options screen.

Default is 1, which gives a reasonably degree of power saving, with hopefully no hugely apparent lost of performance.
Level 0, turns off any power saving, but due to come other core changes, this version always takes less current that the last Stable release on 19th Jan.
Maximum level is 10, there may be noticeable degrading of responsiveness to Rx signals

rogerclarkmelbourne, 2021-02-07:
There were some bugs with the hotspot in the previous binaries I posted.

This is an update which fixes those problems

Also, the number of levels has been reduced from 10 to 4.

Level 4 is equivalent to Level 8 in the previously posted version

[OpenGD77_Eco_levels_202102080837.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/5940062/OpenGD77_Eco_levels_202102080837.zip)

vk7js, 2021-02-07:
Hi Roger,

 

I just tried this version but my radio came up without voice prompts so I had to go back to the prior version with 10 power levels.

Joe

vk7js, 2021-02-08:
Hi Roger,

 

I’m very impressed. I’m using the version with the 10 power saving levels due to voice prompts not working on the later version, but at level 3, after a usual day’s usage, my battery is at 86% It is usually at 50%.

Joe

 

Vk7js

rogerclarkmelbourne, 2021-02-13:
https://github.com/rogerclarkmelbourne/OpenGD77/releases/tag/D2021.02.13.01

closed 2021-02-13 by rogerclarkmelbourne