rogerclarkmelbourne/OpenGD77 #7: Localisation (non english language support)

rogerclarkmelbourne, 2019-11-01:
Currently OpenGD77 is only available in the English language

In the future multiple other languages could be supported, either by different versions of the firmware for each language, or perhaps by storing all the localisation text strings in the Flash memory.

Some changes to the code would be required

Or alternatively a version that would require a rebuild and probably easier to implement would be to have a "strings" array in the code and all text that needs to be localised would need to reference the localisation strings array

lasantiago, 2019-11-04:
Hi Roger, I would be happy to help with this for (at least) Spanish

rogerclarkmelbourne, 2019-11-04:
Ok.

Perhaps I can slowly modify each screen to use a separate array of strings.

Can you tell me whether things like “TG” need to be translated ? ( I guess so )
What about “R” and “T”  freq on the VFO ?

Perhaps you can make a CSV file of all the English text and the Spanish equivalent

f1rmb, 2019-12-27:
I think this one could be closed now.

rogerclarkmelbourne, 2019-12-27:
Yes. Definitely ;-)

closed 2019-12-27 by rogerclarkmelbourne