rogerclarkmelbourne/OpenGD77 #8: Need to optionally filter on currently selected Tx TG, TS and CCode

rogerclarkmelbourne, 2019-11-01:
Currently the firmware effectively runs in "Digital Monitor" mode, where the radio will hear all TG's on the current frequency.

There needs to be an option to enable / disable "Digital Monitor" mode, and this will need to compare the current Tx TG, with the received TG, and only unmute the speaker if there is a match.

Currently the radio does not filter or mute by Timeslot either, so it will need to filter by the current TS when not in Digital Monitor mode.

It may also be possible to filter or not filter by Color Code. Currently the firmware does filter by CCode, however it appears that it may be possible to also disable CCode filtering.
But doing so may cause some DMR audio artifacts, as it makes it more difficult to determine whether a valid DMR signal is being received if the CCode value is not part of the validity check.

rogerclarkmelbourne, 2019-12-06:
Colour Code filtering can't be removed as its part of the HR-C6000 internal functions (or appears to be).

TS filtering is now optional.

TG filtering still needs to be added

Teeed, 2020-02-04:
Well, looks like #420, but it needs some minor improvements.

closed 2020-02-14 by rogerclarkmelbourne

rogerclarkmelbourne, 2020-10-04:
This was implemented a few months ago