rogerclarkmelbourne/OpenGD77 #9: Support for sending and receiving "messages"

rogerclarkmelbourne, 2019-11-01:
Currently the firmware does not support sending or receiving messages.

From what I understand from reading the datasheet on the HR-C6000 DMR DSP, which handles the DMR data, the way that the HR-C6000 needs to be controlled when sending and receiving messages is different from sending and receiving voice calls.

So I suspect a new set of states will need to be added to the state machine in fw_HR-C6000.c and new code written to handle this

Additionally the menu system etc will need new screens to allow the creation and viewing of messages, and they will also need to be stored somewhere in the EEPROM.

Scyten, 2019-12-20:
Hi, is there an idea how to implementieren this? How to distinguish Audio or Data/Messges?
If you provide the technical stuff others may implement the UI.
Yours

alexernst, 2019-12-20:
At the moment the underlying DMR code does not handle data frames others than needed for voice transmission. 

I'm investigating what's needed to do this. But the available documents about the C6000 DMR chip is not very detailed and reliable. So a lot is trial and error.

rogerclarkmelbourne, 2019-12-20:
During the Christmas holiday, I hope to have time to solder some wires to the SPI bus between the MCU and the C6000 so that I can capture the traffic using the official firmware.

I should be able to send a text message and capture that data.

However its not going to be easy attaching the wires.

alexernst, 2019-12-20:
I already captured a data header in a test firmware.
The difference to sending and receiving data frames does not seem to be too big. 

The ReceiveDataInterrupt has to get a state machine to differentiate the packages and assemble received messages and send the needed ACK and NACK packages.

There is some information missing in the docs. But together with the ETSI documentation I can make some knowledge-based guesses.

So for the moment no need for microsurgery :-)
At least not for this issue.

rogerclarkmelbourne, 2019-12-20:
OK.

I have done some captures using GDB but did not find them very useful.

rogerclarkmelbourne, 2020-04-30:
I'm closing all enhancement issues

rogerclarkmelbourne, 2020-10-04:
Re-opening as part of issues cleanup

Scyten, 2021-02-13:
@alexernst  wrote me, that he has implemented a rudimantary mssage functionality and send you an e-mail since the code is not updated in this repo. He does not got an answer to his mailed "pull-request". May you can look for this specific mail. The mail must be before the 21.12.2020

rogerclarkmelbourne, 2021-02-13:
The last time I discussed this with Alex, he was still developing the message function and it was not finished.

He will contact me when he has finished this and perhaps wants to share his code.

alexernst, 2021-02-13:
I have some success sending and receiving messages. But it is in a very early state. The source code is not really usable right now. 
As the documentation of the C6000 is somewhat 'cryptic' and not complete, it's a lot of guessing and trying.

And to add some more trouble repeater and hotspot behave different. Even repeater from different manufactures do not do the same things. In theory everything should be done as defined in the ETSI docs. But this is not as it is.

So still a lot work to do. And my time is currently limited.