rogerclarkmelbourne/OpenGD77 #12: Memory corruption may be caused by the codec binary functions

rogerclarkmelbourne, 2019-11-02:
The DMR AMBE codec used in the firmware comes from the parts of the official firmware (v 3.1.1 ) binary. Hence it operates as a "black box".

Kai did some analysis on the memory usage of the codec and has allocated the entire lower bank of RAM (64k) to the codec, by not allowing the linker to use that bank of memory.

I did some analysis and only around 8k in the middle of the ram is used.

The attached file is a memory dump of the lower 64k bank of RAM, which was initially filled with 0x55 and the radio was then used for Tx and Rx for around half an hour.

Hence it looks like the only part of this bank of RAM which the codec uses is addresses 0x6B60 to 0x8D88

Hence it would be possible to modify the linker files to allow use of other parts of the lower bank of RAM by the OpenGD77 firmware if the firmware ran out of RAM (which currently is not a problem).

**However..**
Its possible that the codec also uses memory in the upper 64k bank of RAM, but its not that easy to determine whether the codec uses RAM in this bank as the bank is used by the OpenGD77 firmware itself. 

One method to find out whether the codec uses the upper bank, would be to change the linker so that the sections in the lower 64k of RAM that are not used by the codec are used by the OpenGD77 firmware, leaving the upper bank of ram unused.

Then run the radio for a while, and dump the upper bank of RAM

[OpenGD77_lower_64k_memory_bank_dump.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/3800091/OpenGD77_lower_64k_memory_bank_dump.zip)

alexernst, 2020-01-10:
I did change the linkerfiles to only use the lower ram.
This works for all variables with one exception: moving the ucHeap used by FreeRTOS to a memory section in lower ram breaks DMR audio output.
If I move the ucHeap to lower ram receiving a DMR qso results in white noise output through the speaker. This does not effect transmitting in DMR or analog at all.

The ucHeap is a 20K memory block used by FreeRTOS.
I'm not sure right now, if this is a problem of the codec. It might also be some problem transfering the decoded audio.

alexernst, 2020-01-10:
After some memory dumps, it doesn't look like the AMBE-codec uses the upper RAM.

rogerclarkmelbourne, 2020-01-10:
Ok.

Thanks

closed 2020-01-10 by rogerclarkmelbourne