rogerclarkmelbourne/OpenGD77 #17: Dual watch

rogerclarkmelbourne, 2019-11-04:
Several people have requested Dual Watch functionality.

I wonder if it may be better to implement some sort of Zone Watch feature, so that more than 2 channels can be watched.

I'm not sure how this differs from the scanning function, which also needs to be added.

In Dual Watch I presume both channels are monitored and the radio locks onto whichever channel has a signal.

In Scanning, I think perhaps the radio only stays on the frequency for a short while before continuing to scan, but I'd need to double check.

Potentially scanning and Dual or Zone watch are almost the same functionality, except have different optional parameters in terms of what happens when a signal is received.

footloosejava, 2019-11-28:
+1  for Zone Watch

rogerclarkmelbourne, 2020-04-30:
I'm closing all enhancement issues

rogerclarkmelbourne, 2020-10-04:
Re-opened as part of issues cleanup

ssoespre, 2020-11-20:
Dual Watch feature is awesome on Development Release of 12 November 2020. 
Perhaps consider closing this issue / enhancement ?
For me the dual watch feature works fine (y)

Thank you for all these nice improvements :)

f1rmb, 2021-01-24:
This one has been implemented.

rogerclarkmelbourne, 2021-01-24:
Yes and no...

Ideally we should have Dual Watch on the Channel screen, perhaps by scanning the first 2 channels in the current Zone, but I'm not sure of how we could do the UI etc for that

closed 2021-04-20 by rogerclarkmelbourne