rogerclarkmelbourne/OpenGD77 #18: Scanning

rogerclarkmelbourne, 2019-11-04:
Several people have requested the Scan feature in the official firmware.

However the official Scan feature is AFIK not ideal, and the Scan feature probably needs to be designed with Amateur radio operators in mind.

The Scan and the Dual Watch functionality seem very similar and potentially could be combined, but some options may need to be added to the Scan to enable this e.g. Scan hold time of indefinite when on a signal.

Also in the Scan, I know people wanted to be able to lock onto a channel even if the signal ended.

Potentially the radio could be made into a more general purpose scanner, in the VFO where the start and end frequency could be set as well as a step etc.

Limitless possibilities ;-)

ajorg, 2019-11-15:
Something I've done for other radios is download the full database of coordinated repeaters for my area (in my case from [WWARA](https://www.wwara.org/)) and pull that into my radio in [a format it understands](https://wwara.s3-us-west-2.amazonaws.com/ICOM/wwara-icom.csv). Then instead of scanning VFO-style where I don't necessarily know the correct offset or tone to access a repeater, I scan through that list of repeaters.

Speaking of which, I'd love to get away from the channel and zone oriented data model DMR radios use and back to the repeater and talkgroup data model hams actually use.

rogerclarkmelbourne, 2019-11-15:
@ajorg 

This firmware does move away from the channel / zone method that is needed for traditional / commercial DMR radios

You only need 1 channel per frequency.

TG's and TS's on that Channel are defined using the Rx Group list - which is now just a Talkgroup list

The latest CPS and firmware also allows TS's to be assigned to the TG for each Digital Contact

rogerclarkmelbourne, 2019-11-15:
BTW. 

Probably the simplest scanning system is to implement a Zone Scan, using the Quick menu

uiChannel mode already stores the index into the current zone, so you could easily keep stepping up though the channels in the zone and stop when a signal is received.

And perhaps use the Function key to exit from Zone Scan

Probably need to write some function in fw_trx.c like  bool trxIsReceivingSignal() which does different checks on DMR to Analogue to determine if there is a signal on the current freq

Scyten, 2019-11-27:
I like the zone scan, that is implemented now!

Is it possible to implement this in VFO, too?
RX & TX: frequency_{n+1} = frequency_n + stepSize.

May be two modes upcounting or downcountig.
Nice to have: with configurable frequency boundaries.

Just as an idea. 

By the way: All of you, especially Roger, are really doing a great job!

rogerclarkmelbourne, 2019-11-27:
@Scyten 

I think Colin G4EML is continuing to work on the Scan functions and is implementing Scan Options at the moment

I agree VFO scan between Rx and Tx frequency is probably the best user interface for this

The other option is to use VFO A Rx ----> VFO B Rx as the range, but its more difficult to change VFO (via the Orange button to select the active VFO), than to change the Rx and Tx frequencies

Also for VFO scanning, it would be best if Colour Code filtering can be disabled, however I have not managed to make this work correctly yet.

closed 2019-12-21 by rogerclarkmelbourne