rogerclarkmelbourne/OpenGD77 #21: Display corruption

rogerclarkmelbourne, 2019-11-06:
Display seems to suffer from occasional corruption, mainly on the Tx screen.

I don't know whats causing this.

The display data is bit-banged using the MCU to generate both the clock and the data signal, because the display is not connected to pins on the MCU which support SPI.

So potentially any part of the data transfer to the screen could get interrupted by an ISR running, which would cause the clock or other signals to have variable length.

The datasheet does not indicate that there are any maximum times for any of the signals, including the clock, data, CS or CD signals.

The CS is held low the whole time, however I did some tests only enabling CS when writing to the IC and it didn't seem to help with the display corruption, so the problem does not seem to be noise on the data when the display is not actively being updated.

One cause could be that the display is being clocked too fast.
The minimum total clock cycle time is 80ns, which is around 10 clock cycles on the MK22, which runs at 120Mhz, however adding NOP's to delay the cycle time didnt seem to make any difference, and its important not to slow the display transfer down too much as the display is normally rendered completely, i.e 1k of data is transferred, which will take around 0.75mS.

rogerclarkmelbourne, 2020-01-20:
I tried modifying the code to check whether a C6000 ISR had been run during the display render function and if so to re-render, but this didn't seem to solve the problem.

f1rmb, 2020-01-20:
@rogerclarkmelbourne can you share the patch ? I would like to test it and compare to what I tried to do about this.

rogerclarkmelbourne, 2020-01-20:
I just set a flag in the master ISR handler in fw_hr-c6000.c and in ucRender rows I wrapped the code in a do{} while() loop

I set the flag to false at the top of the do while loop, and check in the   while   that the flag is still reset

rogerclarkmelbourne, 2020-04-02:
Seems to have been fixed by me moving the software SPI to a different file and disabling compiler optimisation on that file

closed 2020-04-02 by rogerclarkmelbourne