rogerclarkmelbourne/OpenGD77 #27: Battery percentage does not get updated when there is no other change to the display

rogerclarkmelbourne, 2019-11-07:
The header row needs to be rendered if the battery percentage changes when the radio is idle, otherwise the battery percentage will not change if the battery voltage changes e.g. if the radio is in the charger.

Perhaps add a variable to menuQSOData.c rather than in both the VFO and Channel mode screens

rogerclarkmelbourne, 2019-11-29:
I'm closing this as I think it is updating

closed 2019-11-29 by rogerclarkmelbourne

khaytsus, 2020-01-16:
@rogerclarkmelbourne I've noticed this myself the last few days and I'm pretty sure this is not fixed, twice I've seen it by kind of coincidence watching the battery drop 15-20% at a button press and realizing I think it was just not updating, and today I had the radio on for about 2 hours and it was still at 100% and dropped to 96% as soon as I touched a button.