rogerclarkmelbourne/OpenGD77 #28: fw_SPI_Flash.c  Write problem when optimisation is enabled

rogerclarkmelbourne, 2019-11-07:
fw_SPI_Flash.c Is currently has compiler optimisation disabled, because if it’s enabled SPI Flash Write does not work correctly on all radios.

The interface to the chip is bit-banged, so some delay e.g. ASM NOP is probably required in the Write or Page Erase functions

rogerclarkmelbourne, 2020-02-14:
The code seems to not generate clean SPI signals when optimised, so its best not to optimise this file.

In the longer term the bit-banged SPI probably needs to be written in assembler.

rogerclarkmelbourne, 2020-04-02:
Software SPI needs to be moved into a separate file that does not have optimisation enabled

rogerclarkmelbourne, 2020-04-30:
I'm closing all enhancement issues

rogerclarkmelbourne, 2020-10-04:
code in static uint8_t spi_flash_transfer(uint8_t c) and other optimisation sensitive code should probably be moved to a separate file, like had to be done for the UC1701 with UC1701_transfer.c

So that just the optimisation sensitive code gets optimised

f1rmb, 2021-03-06:
I guess you nailed that problem ;-)

closed 2021-04-20 by rogerclarkmelbourne