rogerclarkmelbourne/OpenGD77 #30: DTMF and Tone deviation is incorrect

rogerclarkmelbourne, 2019-11-08:
The DTMF and Tone, FM deviation seems to be too much.

I checked the calibration code and only the value(s) for the Voice deviation are being read from the calibration data, and applied to bits 6 to 15 of register 0x59

In the AT1846S data sheet, bits 0 to 5 of register control the CTCSS and CDCSS deviation, but I can't find any information about setting the deviation for DFTMF and Tone transmissions.

Its possible that the same deviation control bits for CTCSS / CDCSS also control the DTMF and Tone deviation, but this will need to be confirmed by varying the values in register 0x59 bits 0 to 5 and observing the result when transmitting a tone.

alexernst, 2019-11-08:
According to RDA1846S_register_table.xls 0x59[15:6] is also used for tones.
Also mentioned is 0x41[6:0] voice_gain_tx. The value for transmitting tones seems to be 5. 

During initialisation of A1846S 0x41 is set multiple times to different values.
These  values are probably used in the stock firmware.

rogerclarkmelbourne, 2019-11-08:
Ok.

I have some photos of the waveforms, sent to me by VK3YNV and with the official firmware the signal is much cleaner

The calibration data contains values for the DTMF and Tone deviation, but we are not reading theses and applying then to the hardware.

Possibly some radios need less deviation

I can’t find any information on setting the deviation of DTMF and 1750 tone.
But I think perhaps it’s register 0x59 bits 0 to 5, as this is the CTCSS tone deviation and perhaps applies to all tones?

The Arduino Hamshield uses the AT1846, so perhaps there is some information in that repo, about setting DTMF deviation

alexernst, 2019-11-08:
In the docs folder is a file RDA1846S_register_table_20111008.xls containing a sheet 'Tone' which says 0x59[15:6] for FM deviation.
Turning calibration on and off changes the waveform.

If the calibration data contains values for tone deviation I would start with that.
What`s the address of the Tone deviation data values?

rogerclarkmelbourne, 2019-11-08:
http://members.optuszoo.com.au/jason.reilly1/GD-77tune.htm

8F05E | 0F | UHF TX DTMF deviation | Higher value=more deviation
-- | -- | -- | --
8F05F | 0D | UHF TX 1750Hz tone burst deviation | Higher value=more deviation
8F060 | 11 | UHF TX CTCSS deviation, wideband | Higher value=more deviation
8F061 | 11 | UHF TX CTCSS deviation, narrowband | Higher value=more deviation
8F062 | 0B | UHF TX DCS deviation, wideband | Higher value=more deviation
8F063 | 0B | UHF TX DCS deviation, narrowband | Higher value=more deviation

rogerclarkmelbourne, 2019-11-08:
From the "programminguide 1.4" pdf

11. Tx FM Deviation
0aH[10:6] pga_gain<4:0> Voice analog gain.
41H[6:0] voice_gain_tx<6:0> Voice digital gain.
44H[11:8] gain_tx<3:0> Voice digital gain after tx ADC down sample
59H [15:6] xmitter_dev<9:0> Ctcss/cdcss + voice dev setting
59H [5:0] c_dev<5:0> Ctcss/cdcss dev setting
Adjusting 59H [15:6] ( xmitter_dev) can change Tx FM deviation of voice and subaudio.
Adjusting 59H [5:0] ( c_dev) can only change Tx FM deviation of CTCSS and CDCSS

I think its 

59H [5:0] c_dev<5:0> Ctcss/cdcss dev setting

Which is not being adjusted

rogerclarkmelbourne, 2019-11-09:
Hi Alex

I did some tests and I agree with the code you have comitted to your branch, that the setting which needs to be changed is the overall deviation value.

`I2C_AT1846_set_register_with_mask(0x59, 0x003f, xmitter_dev, 6);`

I tried changing the CTCSS in the lower 6 bits of register 0x59 but it didn't change the beep deviation

rogerclarkmelbourne, 2019-11-13:
Fixed

closed 2019-11-13 by rogerclarkmelbourne