rogerclarkmelbourne/OpenGD77 #31: Zone menu not working if zones are not continuously numbered

alexernst, 2019-11-08:
The zone menu is load the wrong zones if pressing up on 'All channel' zone or down on last zone. 

This will only happen with codeplugs where the zone index is not consecutive. eg. I have zones with index 30-45 and 50-53 but not 46-49.

The leads to two problems:
- the number of zones is lower then the highest zone index
- selecting the next or previous zone does not deal with empty zones

rogerclarkmelbourne, 2019-11-08:
Ok.
I could ask Colin G4ELM to look at this problem because he wrote the All channels system, but I think he is too busy and also the code has changed a lot since he wrote the All channels system about 4 months ago.

I will have a look to see if there are any easy solutions

alexernst, 2019-11-11:
No error in OpenGD77. Zones in my codeplug were buggy. Somehow the last zones was duplicated multiple times.

closed 2019-11-11 by alexernst