rogerclarkmelbourne/OpenGD77 #34: DMID database compression

rogerclarkmelbourne, 2019-11-09:
Some other DMR radios compress the DMR ID database by compressing the Callsign and name text etc into 6 bits per byte.

6 bits bits allows 64 characters

0 - 9,A-Z,a-Z, Space and full stop

Any characters outside this range are mapped to the most appropriate character

This functionality would require a change to the CPS CE.

The DMR ID header starts with "ID-V001" which is read by the official firmware to confirm the Flash memory contains valid DMR ID data.

With the OpenGD77 and my modified Official firmware version, I use the letter "V" as a variable to control the record length

The ASCII value of V is 0x56  and from this 0x4A is subtracted in the firmware, which yields 0x0C (12 bytes), which is the standard record length.

Longer record lengths are signalled by using letter codes higher than V (0x56).

To support both compress and uncompressed DMR ID data, the digit "1" in the header (or any of the other digits) "001" could be used to indicate the compression version. 
"1" could indicate normal non-compressed

"2" could indicate 6 bit compression.

ajorg, 2019-11-15:
Is the full stop character required for this scheme? If not, I'd suggest instead letting the two additional characters be ` ` and `-`.

I've pulled the [frequency that characters occur in the DMR ID database](https://gist.github.com/ajorg/dc1890d0ffa86aad9f64), to show that ` `, `.`, and `-` are the most frequent non-letter characters (in that order). However, if you compare [the effect of removing `.`](https://gist.github.com/ajorg/1b49bc11d8561375d30ebe60aa53e2f3/revisions?diff=split) and [the effect of removing `-`](https://gist.github.com/ajorg/926d8c7d2c3afaf946f4ae305e90c42a/revisions?diff=split) it becomes apparent that nearly all `.`s are superfluous, while most `-`s are part of Asian and European names, and may be important to their owners.

rogerclarkmelbourne, 2019-11-15:
I'm not sure.

I think I read about the system used by other manufacturers, possible Connect Systems or Anytone.

But I can't remember what character encoding they used.

rogerclarkmelbourne, 2020-04-30:
I'm closing all enhancement issues

closed 2021-04-20 by rogerclarkmelbourne