rogerclarkmelbourne/OpenGD77 #55: TS change by pressing the Star key is overridden my Digital Contact

rogerclarkmelbourne, 2019-11-11:
Because of the feature that allows Digital Contact (TGs) to specify an override TS.

If the Digital Contact has an override TS, and the operator overrides this override by pressing the Star key.
Then when the PTT is released and the Channel screen is reloaded, the TS will be set to the override specified by the Digital Contact - which will change the TS

The Star key TS change needs to be stored as an override which has higher priority than the Digital Contact TS override

alexernst, 2019-11-12:
There is another problem where the TS is not changed if switching between channels with different TS.

rogerclarkmelbourne, 2019-11-12:
Can you explain

I fixed a problem yesterday, where if the Digital Contract was not set to Override the TS was not being set to the TS for the channel.

I have written a fix for the Star key TS override being reset when the PTT is released and the Channel or VFO screen loads.

However, there is another complication

e.g.

If in the VFO the TS is changed from TS2 to TS1, then the operator switches to a channel where the TS is TS2 or the Digital Contact override is set to TS2.

Should the TS be changed when the Channel screen is loaded, because the operator will probably think that the changes to the VFO are only related to the VFO and switching to the Channel should have the original TS for the Channel

So I have written a second attempt to fix this problem, where I store a separate TS override for both the VFO and the Channel screen (I'm using a uint8_t where the upper nibble is the VFO TS override and the lower nibble is the Channel TS override)

I think perhaps this is the optimum to fix this issue, but perhaps you found another problem.

alexernst, 2019-11-12:
I have two channels, one with TS 1 the other with TS 2. If I switch between the channels the TS stays on 1. 

If I select the first channel, switch to TS 2 and then switch to channel 2 the TS changes to 1 but should be 2.

Tomorrow I will recheck it with a basic codeplug to verify it.

rogerclarkmelbourne, 2019-11-12:
I have pushed my fix for the TS button override to a separate branch

https://github.com/rogerclarkmelbourne/OpenGD77/tree/star-key-ts-override-fix

Can you pull and test it, and tell me whether perhaps it fixes your bug.

I think however that your bug is perhaps not a bug, and that sometimes the Channel data must take priority over the override.

The one occasional when the override has priority is the TG.

At the moment if you manually enter the TG, it stays the same if you change from VFO to Channel and if you change Channel.
The TG only changes if you press Left or Right to select a different TG.
But I think this is probably OK, and no one has a complaint about this functionality.

The TS override I think is less obvious, and hence my change to 
https://github.com/rogerclarkmelbourne/OpenGD77/tree/star-key-ts-override-fix
is perhaps the best compromise

rogerclarkmelbourne, 2019-11-12:
OK.

I think that changes to channel should reset he override on TS caused by the Star key

Also if you have specified a TS override on a Digital Contact, and you press Left or Right to change TG, it should change TS to the TS override vale for the Digital Contact.

The TS override with the Star key should only be temporary if you don't change anything else.

rogerclarkmelbourne, 2019-11-12:
A few people have tested the fix and it seems to be working, so I've merged the branch into the master

alexernst, 2019-11-12:
I did a quick check and my problem persists.

But I will start with a clean codeplug as that might be a problem. 

Will report later.

rogerclarkmelbourne, 2019-11-12:
Ok

Thanks

alexernst, 2019-11-13:
Tried the simple default codeplug of the official CPS, changing only TG on one channel.
That works like I would expect. 

Looking at my codeplug I found that all contacts have channel TS override set to TS1.
Don't know how this happens.

So no problem anymore. And I will start with a new empty codeplug.

BTW:
If I load my codeplug into the community CPS the last zone shows up multiple times.

rogerclarkmelbourne, 2019-11-13:
Hi Alex

OK.

By default the "reserve1" byte in the Contact is set to 0xff in the ContactOne constructor

However some codeplug tools sometimes set it to 0x00.

I'm using Bit 0 of reserve1 as a flag. If bit 1 = 0, then the contact is set to override.
Then Bit 1 contains the TS,   i.e 0 = TS1 1 = TS2

In your case I think reserver1 has somehow become set to 0x00

I have checked various codeplugs that people sent me and they all seem OK, with the reserve1 byte set to 0xff

rogerclarkmelbourne, 2019-11-13:
Fixed

closed 2019-11-13 by rogerclarkmelbourne

alexernst, 2019-11-13:
I've found the explanation.

I used [https://github.com/sergev/dmrconfig](dmrconfig) to set all channels, contacts, zones, ... in my codeplug. And dmrconfig simple sets unused bits to 0.

So all contacts have the TS1 override set.

rogerclarkmelbourne, 2019-11-13:
Can you raise an issue for that tool, as the last byte should be set to 0xFF

rogerclarkmelbourne, 2019-11-13:
Btw.

Colin G4ELM has now updated his tools to export and import the last byte, as previously he didn’t export it, and when he imported , his utility set the byte to 0xFF, which removed all overrides

rogerclarkmelbourne, 2019-11-13:
I've raised an issue on that project

https://github.com/sergev/dmrconfig/issues/43

The minimum they should to is to use the correct default value of 0xFF