rogerclarkmelbourne/OpenGD77 #57: In FM mode, screen backlight does not go off while signal is being received

rogerclarkmelbourne, 2019-11-12:
The screen backlight should illuminate when a FM signal is first received, but should go off again depending on the backlight timeout setting

alexernst, 2019-11-13:
Perhaps we should add this as an option to the display options menu.
I would like it the other way round: Keep backlight on while signal is received in DMR and go off after end of signal + backlight timeout.

rogerclarkmelbourne, 2019-11-13:
Yes.

That is also a good idea.

I normally leave my radio in the charger and have the backlight on all the time.

But it would be better if the backlight could stay on while a call is being received.

This would probably apply to both FM and DMR, but its more important for DMR because of the ID or callsign + name display

rogerclarkmelbourne, 2019-11-13:
Fix is currently being tested....

closed 2019-11-13 by rogerclarkmelbourne