rogerclarkmelbourne/OpenGD77 #61: Promiscuous Mode over both Timeslots.

DL7PJ, 2019-11-13:
in the old firmware I always have promiscuous mode over both timeslots. With OpenGD-77, I only see the LED green light but no audio from the other timeslot.

rogerclarkmelbourne, 2019-11-13:
With Tier2 you must select a TS.

Please propose how the UI to support this

DL7PJ, 2019-11-13:
the old FW has dual VFO. My setup was VFO1 TS 1 and VFO2 TS2 and my hotspot is sending TS: 1 TG262112 + TS1 TG26322 and TS2: TG2625 + TG26256. So I was able to monitor 4 Talk Groups

DL7PJ, 2019-11-13:
example photo for two timeslots :
[image]

dc7jzb, 2019-11-16:
PJ, now I know what you mean. 8)

Roger, he has worked with the double wait function in the stock Firmware, so the Radio was Jumping between VFO A and B. And on A he had one TS on B the other.
So I think you can take this as a feature request for a Dual Watch function. ;)

DL7PJ, 2019-11-18:
thank you Mario, that's exactly what I meant and miss in the OPENGD77 FW
Roger that would be optimal. Thank you for the great FW and great work

rogerclarkmelbourne, 2019-11-19:
Its on the To Do list as part of the general DMR Filtering

I'll should open another topic on the forum as currently its under the Dual Watch thread

http://opengd77.com/viewtopic.php?f=4&t=4

rogerclarkmelbourne, 2019-11-27:
This has now been implemented

closed 2019-11-27 by rogerclarkmelbourne

DL7PJ, 2019-11-27:
Hi Roger , I have tested, great job works very fine and switching is very easy , this is exactly my wish , thanks and 73 Peter 
--
Gesendet über myMail für Android Mittwoch,

rogerclarkmelbourne, 2019-11-27:
No worries....