rogerclarkmelbourne/OpenGD77 #66: RX Not locking?

G0FHM, 2019-11-14:
RX appears to not be locking up at 9MHz frequency split (Most UK DMR repeaters).  Tested at 3MHz split, all ok

rogerclarkmelbourne, 2019-11-14:
Which specific repeater an frequency

G0FHM, 2019-11-14:
GB7GS - 430.450 (in) 439.450 (out). Radio TX is asking the repeater to
activate the downlink (which is what it does) but the RX is dead. Not even
the status LED is on. I've messed around a bit, and it works ok with  3MHz
split, so i'm assuming the wider split is confusing something.

Thanks

rogerclarkmelbourne, 2019-11-14:
Ok.

I will configure my duplex Hs to the freq and test

rogerclarkmelbourne, 2019-11-14:
Just to confirm
The repeater listens on 430.450 and transmits on 439.450 ?

G0FHM, 2019-11-14:
Confirm

rogerclarkmelbourne, 2019-11-14:
Strange.

It looks like the Tx and Rx freqs are being set correctly. 

But at the moment my duplex hotspot is not playing ball. I've just realised its not working at all :-(

I'll need to fix it before I can do any more tests

G0FHM, 2019-11-14:
No problem. I'd put it down to a dodgy radio, but I have 2x GD77 doing the
same!

I'll sit tight in the absence of a working duplex board

Thanks,

J

rogerclarkmelbourne, 2019-11-14:
No worries. 

I'll try to replicate the bug as soon as I can get my HS working

vk7hse, 2019-11-15:
Hi Roger,

Clayton has forwarded me a copy of this firmware to test on one of my duplex hotspots. I'll give you a call once I have it flashed and (hopefully) all working!

rogerclarkmelbourne, 2019-11-15:
Thanks.

For some reason my duplex hotspot was not working this morning, and I've not had chance to test this problem. 

I know the firmware was working on duplex hotspots a week or two ago, and no one has contacted me to tell me anything different.

But there have been some changes over the last few days associated with frequency resolution previously the freq could not be set to the nearest 100Hz, but @alexernst noticed that 6.25Khz step's were not working as it requires 50Hz resolution, so he added another decimal place to the freq handling and there is a chance that something got missed in that change which has caused this bug.

Or it could have been here for months ;-)

vk7hse, 2019-11-15:
No worries, my guess was going to be a frequency indifference. As you know some of the these cheep duplex devices are not very stable in their frequency lock.

Anyway I'll report back asap... ??

rogerclarkmelbourne, 2019-11-15:
Thanks

My HS seems to be working again now. I think it wasn't working because the HS Dual hat needs a -200 offset, but I'm running the same RPi as when I was testing the GD-77 hotspot which needs an offset of 0 because its much more accurate.

rogerclarkmelbourne, 2019-11-15:
I've now managed to set my HS up on the same frequency as that repeater and it works fine.

@G0FHM what firmware version are you running, please post the link to where you downloaded it from

G0FHM, 2019-11-15:
Here is the download link:

https://github.com/rogerclarkmelbourne/OpenGD77/tree/master/firmware_binaries/daily_builds

I have a second 77 here, I'll flash that and see what happens

G0FHM, 2019-11-15:
Just in case it will help, a short video of what's going on:

https://youtu.be/e9c1Qu424Jk

rogerclarkmelbourne, 2019-11-15:
Do you mean

https://github.com/rogerclarkmelbourne/OpenGD77/raw/master/firmware_binaries/daily_builds/OpenGD77_Tier2_latest.sgl ??

rogerclarkmelbourne, 2019-11-15:
Backup your EEPROM, Flash and codeplug and post them here in a zip file

PS What colour code is the repeater on ?

rogerclarkmelbourne, 2019-11-15:
Also, are you using a channel or the VFO ?

And, can you try doing a factory reset

Hold Blue button in while powering on the radio.

vk7hse, 2019-11-15:
Initial testing on my brandmeister duplex hotspot using the radio in VFO mode worked just fine. Now the only difference is that for ease of making a quick test I didn't alter the 7MHz split I'm running on the device. Sometimes these duplex hats require a longer handshake period, on stock firmware I've got mine set to 1024ms with the maximum number of retries. Local interference (no cavities filter to prevent the TX overwhelming the RX) will lead to failure to keyup every time especially if that handshake time is too short (think the stock firmware is set at or around 320ms with only one retry)

The only thing I did manage was to cause the GD-77 to reboot when attempting to use a private call (VFO mode) to 9990.

I'll do some more thorough checks tomorrow as well as making a zone so I can test it as a channel instead of the VFO.

Cheers

G0FHM, 2019-11-15:
Repeater CC is 10.

Using a channel as opposed to VFO. Have tried the factory reset.
EEPROM/FLASH in attached zip file.

HTH

rogerclarkmelbourne, 2019-11-15:
@vk7hse 
If you have a channel assigned as a hotspot, you can pull the VFO into the channel

Go into the channel , press the orange button and select option VFO ---> Channel  then press green.

rogerclarkmelbourne, 2019-11-15:
@G0FHM 

I cant see an attached file :-(

G0FHM, 2019-11-15:
As in it's not there? Or won't open?

rogerclarkmelbourne, 2019-11-15:
I can't see it at all
Can you put it somewhere else e.g. google drive etc with a publicly accessible link

G0FHM, 2019-11-15:
Here we go

https://drive.google.com/open?id=1pq_O84nQBflDopuBfBWyEsP9ofLEJo93

rogerclarkmelbourne, 2019-11-15:
Thanks but it didnt seem to include your codeplug

I can extract it from the flash and EEPROM but its a pain.

Can you somehow send the codeplug and also tell me which channel in the codeplug you were using

PS. Its 10pm here so its getting too late for me to do much more, however I'll take a look at the codeplug etc tomorrow

G0FHM, 2019-11-15:
Sorry, forgot about the codeplug!

Only have 3 channels programmed, as I started a new CP from scratch, just
in case.

https://drive.google.com/open?id=1VOtx2S3xocpQkYFP5NFad9mnhkvOj8mH

rogerclarkmelbourne, 2019-11-15:
Ok...

rogerclarkmelbourne, 2019-11-15:
I loaded your codeplug, and it worked fine on my duplex hotspot

When you trigger the repeater using another radio, put the GD77 into FM mode (Function + Star) and it should be able to hear it

Also, BTW. Your VFO A data in the codeplug is out of band, its set to 400Mhz FM

In the channel screen, press the orange button, then copy the current DMR channel to the VFO

Then in the VFO press Function + Star to switch to FM

If you can hear the repeater on FM, press Function + Star to go back to DMR and see if it still picks it up.

rogerclarkmelbourne, 2019-11-15:
BTW.
I had quick look at the calibration data and it visually seemed OK. 
Power settings look like they are also OK

G0FHM, 2019-11-16:
Ok, here is where we are at. Have set both VFO freqs in CPS to 435MHz.

In FM mode, the repeater comes through when it is bought up. Switch to DMR,
the RX goes off, as well as the green LED. If I switch back to FM with the
repeater still transmitting, the green LED comes back however the squelch
doesn't open, even though the RSSI bar graph is showing full strength. When
the carrier from the repeater drops and restarts, the FM signal will come
back again.

rogerclarkmelbourne, 2019-11-16:
This looks like the classic symtoms of the incorrect colour code, but I just checked the repeater listing and its supposed to be on CC10

I tested that CC10 was working but I think I'll need to perhaps to a test build for you with CC filtering disabled

Again, its too late for me to do this now at 10pm  I'll do it tomorrow

BTW. I know its a pain, but you could try setting the other CC values.
Go into the Channel Details screen (Function + Green ) and you can change it

G0FHM, 2019-11-16:
Ok, i'll give that a bash. I'm assuming that CC0 is the equivalent of "open
channel"?

rogerclarkmelbourne, 2019-11-16:
No. I think CC0 is actually a valid code, but I would need to check.

If possible can you register in the new forum  www.opengd77.com and post this bug, because it would be good if someone else in range of the same repeater could replicate this bug

rogerclarkmelbourne, 2019-12-21:
@G0FHM 

Can you try the latest version, as this may have been related to DMR 4FSK deviation calibration problems

G0FHM, 2019-12-22:
That has done the trick, thank you!

rogerclarkmelbourne, 2019-12-22:
OK.

FYI. The problem was not the frequency shift, but would have been related to the frequency of operation.

closed 2019-12-22 by rogerclarkmelbourne