rogerclarkmelbourne/OpenGD77 #70: FM Narrow on Channel Display

ajorg, 2019-11-15:
If my radio is in narrow FM mode, it would be nice to have the display show `FMN` instead of `FM`.

I also think it makes slightly more sense to treat narrowband as a mode, since DMR doesn't let you choose your bandwidth anyway, but I don't see that as especially important, particularly given that the data isn't modeled that way.

rogerclarkmelbourne, 2019-11-15:
There is space for this.

Perhaps 

FM N

Rather than FMN

It’s a very easy change

ajorg, 2019-11-15:
I would not mind trying my hand at contributing that, then. Will take me some effort to setup a dev environment I'm sure, but I'll get around to it.

rogerclarkmelbourne, 2019-11-15:
You could probably almost do this without needing the dev env ;-)

Display for FM is here 

https://github.com/rogerclarkmelbourne/OpenGD77/blob/master/firmware/source/user_interface/uiUtilityQSOData.c#L448-L462

You can get the bandwidth by calling

https://github.com/rogerclarkmelbourne/OpenGD77/blob/master/firmware/source/functions/fw_trx.c#L84-L88

ajorg, 2019-11-17:
I managed to get the dev environment set up, so this has been tested as well. Shows an `N` in 12.5kHz and doesn't in 25kHz.

ajorg, 2019-11-17:
I do wonder if `FMN` rather than `FM N` would be better, so that it can show `FMN CTR 250mW` instead of `FM N CTR250mW`. A matter of taste, I'm sure, so I'm happy to leave it up to the maintainers.

rogerclarkmelbourne, 2019-11-17:
OK
I wasn't sure what would fit on the screen. 
I'm happy to try FMN and see what comments we get

closed 2019-11-17 by rogerclarkmelbourne

rogerclarkmelbourne, 2019-11-17:
BTW.
I've updated the User guide, (but not any photos) to describe this new display functionality

ajorg, 2019-11-17:
Cool!

commit 7e83c6ab (2019-11-17): Add `N` after `FM` when using 12.5kHz FM

Fixes #70

If the radio is in narrowband, both receive and transmit tones are set,
and power is set to less than 1W, the `CTR` still has a space after it.

`FMN CTR 250mW`
PR #78 (2019-11-17): Add `N` after `FM` when using 12.5kHz FM
Fixes #70

If the radio is in narrowband, both receive and transmit tones are set,
and power is set to less than 1W, the `CTR` will be right next to the power.

`FM N CTR250mW`