rogerclarkmelbourne/OpenGD77 #72: Add VFO B support ??

rogerclarkmelbourne, 2019-11-15:
Is it worth adding support for VFO B as an additional screen.

i.e Pressing Red button would cycle through Channel , VFO A, VFO B ??

The CPS has data for 2 VFO's (A and B), and it would not be that hard to add this to the firmware

However, I'm not sure whether it would ever be used.

ajorg, 2019-11-16:
One thing this could help with is having a different offset for each VFO (0.6 for VHF and 5 for UHF) but I'm not sure two VFO channels is the best way to manage that, particularly since the direction of the offset is going to vary depending on the frequency range, and there's no global standard for repeater ranges.

rogerclarkmelbourne, 2019-11-16:
Each VFO has both Tx and Rx freq's so its not the quite the same as most HF radios.

So VFO B would be identical to VFO A except it would be completely independant

I think what are asking for is a method to set the Tx / Rx offset, in which case, something in the Quick menu (Orange button) should accomplish this e.g. have setting for Offset and use Right or Left to cycle between a number of different offsets the press Green to apply

rogerclarkmelbourne, 2019-11-26:
Done

closed 2019-11-26 by rogerclarkmelbourne