rogerclarkmelbourne/OpenGD77 #73: Tier 2 - Simplex Hotspot mode disconnecting PiStar during Tx does not terminate Tx

rogerclarkmelbourne, 2019-11-16:
ON4JPS reported that if when in Hotspot mode when the radio is transmitting, the MMDVMHost task on PiStar is terminated or the power to PiStar is removed etc, that the Hotspot continues to transmit.

I have been unable to replicate this bug, either using MMDVMHost on the PC, where I disconnected the USB cable while the Hotspot was transmitting, or when I terminated the MMDVMHost application.

On PiStar (RPi3B+) running a fully updated Pi-Star:3.4.16  , I removed the power

In all instance the Hotspot stopped transmitting when the incoming network data buffer became empty, because the radio had transmitted the all the data.

EA5SW, 2019-11-16:
I understand that hotspot mode are not available at this time in the released Tier2 Alpha version.
It's correct ?? I tried to test and don't work for me from the Tier2 Alpha lastest bin firmware.
Thanks.

Jose EA5SW

f1rmb, 2020-01-28:
I think this one could be closed.

rogerclarkmelbourne, 2020-01-28:
OK.

closed 2020-01-28 by rogerclarkmelbourne