rogerclarkmelbourne/OpenGD77 #77: Radio Reboots after several Short 1750Hz Bursts or using CTCSS

dc7jzb, 2019-11-16:
The Radio reboots after sending several short Bursts with 1750 Hz Tone.
Sometimes after 3 Short Bursts (Some Repeater come Back with the Call, after 3 Short Bursts)
Sometimes more. (Longest was 30 sec. of short 1750 Hz Bursts before it reboots)

The same we have with active CTCSS, after 3-4 Times Transmitting the Radio reboots.
I've tested and confirmed it with both of my GD77, one with the last Tier II Version before you have activated the Hotspot, the second with the Version from Yesterday.

One User from our Telegram Group had a reset to System Defaults after Reboot.

rogerclarkmelbourne, 2019-11-16:
OK.
I think this is a more general issue.

I have found that repeated pressing keys can cause reboots.

dc7jzb, 2019-11-16:
This could be, someone on our Group told something like that, but I didn't tried that.
Only with the Tone Bursts, so We thought first that it is a Problem of the tone generating.

rogerclarkmelbourne, 2019-11-16:
I didn't notice it on Tone Burst, and assumed it was only when the radio was doing more complex tasks like switching from DMR < -- > FM etc

BTW. Why do you need to keep pressing the Tone button ?

Is there a way to automate this ? 

Can't the radio sense using the RSSI indication, and automatically send tone until the repeater responds ?
The DMR has to do a similar system, except it sends a 30mS pulse and then waits a for a small time before trying again.

How long does tone need to be transmitted ?

dc7jzb, 2019-11-16:
On one or two fm repeaters you can cause some reaction like the call of the reapeater or your Signal strenght, when you send the tone 3 or 4 times in short pulses.
But I don't think that this is really a needed and used function.
Normally it will only be transmitted one short time to open the repeater.
I've never looked how Long a burst has to be for it.

rogerclarkmelbourne, 2019-11-16:
I think I will add it to the list of possible improvements

rogerclarkmelbourne, 2019-11-17:
I did some tests and technically, this looks like its a problem with the access to the AT1846S chip

I think the I2C bus is locking up somehow, potentially 2 things are attempting to access the I2C bus at the same time. 
Well.. While one part of the firmware is accessing the I2C bus there is probably a RTOS task switch and another part of the firmware attempt to access the I2C bus, possibly to use the EEPROM but more likely to also access the AT1846 RF chip.

rogerclarkmelbourne, 2019-11-20:
Bus clash has been logged.

I will debug this.

rogerclarkmelbourne, 2019-11-21:
Fixed in eef5644e80ee31834db5a4af03c73d0b3791d67b

closed 2019-11-21 by rogerclarkmelbourne