rogerclarkmelbourne/OpenGD77 #80: Feature Request: Backlight Hotkey

dc7jzb, 2019-11-17:
Is there a way, to use a Hotkey for switching the Backlight permanent On or Off ?

If I Use the Radio as Hotspot, I have the Brightness at 0%, so the backlight is Off, because I don't want to see it lighting up when a call Comes.
But sometimes I Need the backlight, so I have to go to the Menu, and set it to a higher Value like 15 or 20 % and backwards after it.
I Think a Hotkey for a quick On/Off Option of the Backlight will be fine.

rogerclarkmelbourne, 2019-11-17:
Set backlight timeout to 0 in the Display options to enable all the time

There is no way to disable it at the moment apart from setting the intensity 

What you want would need another setting in the display options.

Please submit a PR for this

rogerclarkmelbourne, 2019-11-17:
Thinking about this.

I suppose the UI would probably be an option for 

Backlight mode 

e.g. 

Backlight:Auto
Backlight:Manual
Backlight:None

In 
**Auto** the backlight would be turned on when there is a signal etc
**Manual** the backlight would illuminate when a key is pressed
**None** the backlight would never come on

closed 2020-01-13 by rogerclarkmelbourne