rogerclarkmelbourne/OpenGD77 #88: VFO --> Channel doesn't persist

ajorg, 2019-11-18:
If I copy the VFO settings to a Channel using the `VFO --> Channel` quick menu item, it does copy those settings over, but if I then change to another channel and back, the new channel settings are lost.

Looking at [the sources](https://github.com/rogerclarkmelbourne/OpenGD77/blob/51ad7b3a95a345bda5c1c6fe7c7907275b1a48f5/firmware/source/user_interface/uiChannelMode.c#L705), it appears that saving the channel back to the codeplug is not done there. I'm also unsure if it should be (do I necessarily want to save that channel to the codeplug? how else should I indicate that I want to?).

Likely this is supposed to be done with `codeplugChannelSaveDataForIndex`, but that seems to have some bugs I'll also report.

ajorg, 2019-11-18:
Opened #89 for the corruption bug I found while trying to test this.

rogerclarkmelbourne, 2019-11-18:
This is by design, but I suppose should be changed

The only way to make the change permanent is to go to the Channel Details screen just after you import and then use Function + Green to commit it change back to flash

However I agree that people would probably expect the change to immediately be permanent

So the quick menu needs to also call the save function use on the Channel Details screen to write the channel data back into the Flash or EEPROM memory

It should be easy to change

ajorg, 2019-11-18:
I'll be happy to try my hand at this one too. Thanks!

rogerclarkmelbourne, 2019-11-18:
Go for it..

It’s literally just one function call from the Channel Details screen

rogerclarkmelbourne, 2019-11-18:
We should probably change the Channel Details screen at the same time

rogerclarkmelbourne, 2019-11-18:
@ajorg 

I can do this if you want.

Its really easy for me..

ajorg, 2019-11-18:
I'm interested in getting more exposure to the codebase, but I don't want
to make more work for you if you want to take this one.

rogerclarkmelbourne, 2019-11-18:
Its very little work, so I'll do it

rogerclarkmelbourne, 2019-11-18:
Actually there was a quirk with this.

I the settingsCurrentChannelNumber was not always being updated when the channel screen is loaded, which may be another bug that need to be investigated.

But added some code to reload settingsCurrentChannelNumber if it contained -1 which is the VFO channel's value

closed 2019-11-18 by rogerclarkmelbourne