rogerclarkmelbourne/OpenGD77 #102: FM narrow indication FMN is hard to distinguish

rogerclarkmelbourne, 2019-11-19:
Feedback from one operator that the display of FM Narrow is sub optimal

Alternative suggestions were FMn and nFM

rogerclarkmelbourne, 2019-11-19:
It may be better to include bandwidth on the right side of the screen to separate it from the FM text

ajorg, 2019-11-28:
[image]

Is something like this better, with an N or W on the right side?

ajorg, 2019-11-28:
I don't think there's enough room for 12.5kHz, and just 12.5 or 25 would be confusing too.

rogerclarkmelbourne, 2019-11-28:
I think its a bit confusing just to have a "N"

Perhaps move the power indication so that rater than it being centered its always 8 pixels to the left of its current location.  (or some other distance)

Then this would leave more space on the right

ajorg, 2019-11-28:
I think I can make that happen.

rogerclarkmelbourne, 2019-11-28:
What will you put on the right hand side ?  "12.5kHz"  or "25kHz" ?

rogerclarkmelbourne, 2019-11-28:
I did think of "WB" / "NB" but 25kHz is not really "wide"

ajorg, 2019-11-28:
I experimented with "12.5kHz" but there really isn't enough room. There's space for about 21 characters, and `FM CTR250mW12.5kHz99%` is 21 characters and clearly too dense.

The more I try to make this work the more I think `FMN`, `FMn`, `NFM`, or `nFM` really is the better way to go. Mostly because you're right that `WB` is not accurate.

I could go with showing `NB` on the right *only* when using 12.5kHz FM, but I'd rather fit `L` on the right when locked.

Or... icons, which frankly I'm not excited about.

rogerclarkmelbourne, 2019-11-28:
Daniel is currently working on extending the fonts, so you could probably have a special symbol for this (though I don't know what)

f1rmb, 2019-11-28:
What about bold font for wide, and regular for narrow.
Of course, our 8x6 bold isn't bold ATM.

Cheers.

---
73 de Daniel, F1RMB
http://f1rmb.blog.free.fr

ajorg, 2019-11-28:
Honestly, I think `FMN` or `NFM` for the mode on the left is the right solution. But I'm happy to try to implement something else if people would really prefer something else.

I'm used to treating narrowband FM as a separate mode from other radios I've owned.

rogerclarkmelbourne, 2019-11-28:
NFM would make it more obvious

ajorg, 2019-11-29:
I've got patches ready for either of these:

[image]
[image]

Maybe worth noting that at least on *my* radio the LCD is places so that I can only see about half of the `N` unless I have the screen turned a bit away to my left.

PS: Thanks for the screengrab feature. It's excellent!

rogerclarkmelbourne, 2020-04-30:
I'm closing all enhancement issues

closed 2020-04-30 by rogerclarkmelbourne