rogerclarkmelbourne/OpenGD77 #92: Pop at the beginning of most beeps

ajorg, 2019-11-19:
There's a sort of a pop sound at the beginning of beeps. It's most noticeable when the beep volume is set very low (below -12dB). Could be something to do with opening the sound device with a high value in a buffer, or just opening the sound device at all, or the sound device might start reading the buffer at a different location than we'd think. Or something else entirely.

Obviously a low priority.

rogerclarkmelbourne, 2019-11-19:
I think what you may be hearing is a click when the audio amplifier is enabled

Possibly its being enabled when there is a DC voltage on the input.

I have this effect all the time even on the official firmware, if I connect an external speaker.

It happens every time there is a DMR signal. Even if the mic volume is set to zero, I get a distinct click each time the amplifier is enabled

This is probably partially a hardware bug, but we may be able to work around it a bit with the firmware, by potentially only turning on the amplier when we provide some negative or positive bias to the amplifier.

I'll look at the schematic to see if there is anything we can do.

Also, when he gets back from his long holiday, I'll ask Jason VK7ZJA about this , because he is the expert on the GD-77 hardware.

rogerclarkmelbourne, 2019-12-15:
Note. External speakers always click due to a hardware design error

closed 2019-12-15 by rogerclarkmelbourne