rogerclarkmelbourne/OpenGD77 #93: Audio stops on button presses (beep related?)

ajorg, 2019-11-19:
This seems to happen on both FM and DMR. If I'm listening to a call and I press a button, I hear a beep, and then even though the radio indicates that it's receiving, I don't hear the call anymore. I wondered if it just needs to check if there's a signal being received at the end of a beep, so that it can start listening again.

rogerclarkmelbourne, 2019-11-19:
Strange.

Its not happening to me. I just tested on DMR

I'll test on FM later

rogerclarkmelbourne, 2019-11-19:
BTW.
We had this sort of bug ages ago, and it was related to the end of the beep turning off the audio amplifier.

But its strange that you are getting this problem and I'm not

ajorg, 2019-11-20:
I haven't seen this since updating to https://github.com/rogerclarkmelbourne/OpenGD77/commit/343eddbfac4e121959f081e3c2fa03028ca332ab but I want to verify a bit more before we resolve.

I thought I saw a comment on the forum about something similar being fixed recently.

rogerclarkmelbourne, 2019-11-20:
I think I did something on FM which may have broken the beep, but I fixed it again.

Was it DMR or FM ?

ajorg, 2019-11-20:
I thought I saw it on both FM and DMR. I will test again on both when I get a chance.

ajorg, 2019-11-22:
I've not been able to repro with recent builds. Closing this, but may reopen if I see it again.

closed 2019-11-22 by ajorg