rogerclarkmelbourne/OpenGD77 #110: Incorrect TA data for some stations

rogerclarkmelbourne, 2019-11-20:
[image]

f1rmb, 2019-11-20:
Exaclty what I had yesterday evening this "minus" something...
But dumping MMDVMVHost log file confirmed that was right (and the value on the question mark position are 0x04, IIRC.

f1rmb, 2019-11-20:
But what is that A/B at the end ?

rogerclarkmelbourne, 2019-11-20:
I don't know

I presume its in the font ??

f1rmb, 2019-11-20:
Nope

f1rmb, 2019-11-20:
I will PR a small change in space removal code (stupid me), but this A/B in are both from the 8x8 font.

rogerclarkmelbourne, 2019-11-20:
OK

f1rmb, 2019-11-20:
Was it KD5ILG ?

rogerclarkmelbourne, 2019-11-20:
Sorry. I don't know

It was on TG91 about 20 mins ago. I should have listened for their callsign

f1rmb, 2019-11-20:
Okay.

rogerclarkmelbourne, 2019-11-20:
They are not active now on 91

I was busy fixing the scan crash bug

f1rmb, 2019-11-20:
Yeah, it's quite calm, I can even check if the new callsign extractor from TA works :-D

rogerclarkmelbourne, 2019-11-20:
(LOL)

Its what's known as "sods law"

rogerclarkmelbourne, 2019-11-21:
Fixed by e0c0fcb2c294b165e3652d5545a7eabcebb22b5d

closed 2019-11-21 by rogerclarkmelbourne