rogerclarkmelbourne/OpenGD77 #121: Add most recent scanned (Found) channels list.

footloosejava, 2019-11-21:
When scanning a Zone, it would be a bonus feature to have the phone remember some of the last Channels that it locked on to.

So menu SCAN and also menu SCAN & LOG:

Scan: regular scan
Scan & Store: - scan and keep a list of the last XX number of channels found.

Scan keep works just like Scan. However, when it finds a channel and that channel is locked onto for more than X seconds, add to Last Accessed list.

Example: The user can scan a Zone or frequency range, and then have any channels found kept for future access.  Perhaps, the last 16 channels or so.

'Scan Log' then is a permanent Zone that stores the last 16 channels found when scanning.
Then also menu, 'clear Scan Log' to erase log and then user can do a scan to make a fresh log.

rogerclarkmelbourne, 2019-11-22:
Scanning a frequency range is definitely a possibility

Perhaps by using the VFO to scan, and setting the Rx freq to the start freq and the Tx freq to the end freq, it would use the step defined for the VFO.

Re: Log of scanned / found

The channels could be logged, but zones are very small at the moment, only 16 channels, so creation of a zone of perhaps frequencies which have been found would not hold very many.

It would be possible to log the frequencies or channels to the USB Com port, but this cuts off the speaker in the hardware connections :-(

rogerclarkmelbourne, 2020-04-30:
I'm closing all enhancement issues

rogerclarkmelbourne, 2020-10-04:
Re-opened for reconsideration

closed 2021-04-20 by rogerclarkmelbourne