rogerclarkmelbourne/OpenGD77 #122: Beginning of echo response interrupted with morse chime

ajorg, 2019-11-22:
I own an openSPOT2, which has a local echo feature on TG 9999. When I try to use that feature, I hear a morse code chime and I'm presented with an "accept call" query. I'm sure this is a private call feature. I don't know if it's also related to #83. In any case, on a Ham radio, I would not expect to have to *accept* a private call, rather I'd expect all calls, private or otherwise, to be accepted always, or at least I'd expect to find an option somewhere I can change that behavior.

At first I thought the morse might be coming from the openSPOT2, possibly because it has an audible morse identify feature I had turned on, but I was able to verify using another radio that the chime I heard is an incoming private call chime.

If there's agreement it would be a good feature to have an option to accept all calls without prompting, I'd like to see if I can make the change myself and send a pull request.

rogerclarkmelbourne, 2019-11-22:
There is already a similar bug in the issues list, where incoming calls are miss interpreted as an incoming Private call sometimes

ajorg, 2019-11-22:
I agree that's related, but I don't think I want to be prompted to accept a call in the case where the call is indeed a private call.

rogerclarkmelbourne, 2019-11-22:
OK. 
AFIK being prompted is a feature on some high end DMR radios, and it is useful because it allows the user to not immediatly need to respond within a few seconds of the signal ending (which is the situation in the official firmware)

A new option will need to be added to enable or disable this, and if its disabled it will be difficult to return a private call

ajorg, 2019-11-23:
Oh, the accept call thing is to switch to the private call ID so that I can respond?

Some other radios I've used accomplish something similar just by temporarily overriding the destination for some timeout period. I don't know if OpenGD77 does that yet.

I actually don't mind the prompt. What struck me as wrong was the chime playing over the call so that I couldn't actually hear the call audio.

Now I'm wondering if I've misunderstood what's going on. I'm going to try some things and get back to you. I hope I'm not making a nuisance of myself.

rogerclarkmelbourne, 2019-11-23:
The OpenGD77 does not set the Tx TG to the incoming TG like commercial radios do, because this is a bug area of confusion for Amateur Radio operators.

For Amateur Radio operators expect radios to do the things they tell them to do, e.g. if you set Tx on TG123 it should transmit on TG123 and not the last incoming PC or TG,  if you pressed the PTT within a specific amount of time.

One of the work-arounds that people use in the official firmware on the GD-77 and most other radios, is to have loads of different Rx groups, each with just one TG in them, so they are only listening on one TG and hence the radio will only transmit on that TG (assuming you set the Tx Contact to the same TG as in the Rx Group list

With the OpenGD77, things are done deliberately differently.

It transmits on the TG you set it to.

Currently TG filtering is not implemented, only TS and CC filtering, and so that people know that the incoming call is not on the same TG that they are set to transmit on, the TG is displayed in inverse video
If the operator wants to transmit on the incoming TG, or PC, they can press the Function key while receiving the signal / call and the Tx TG will be changed to match the Rx TG.

BTW.
There is a bug in the Private Call announcement, but I have not had time to track down whats going on.

ajorg, 2019-11-23:
Ah, that makes a lot of sense. I really like that feature now that I know about it. Knowing that I can use the function key to snag the TG or PC ID, I think I'd rather not have an "accept call dialog", or at least have the option not to.

I also tried to do some private calls on simplex without much luck, but private calls are generally discouraged in Ham anyway since many radios won't let you listen in on calls not to yourself.

rogerclarkmelbourne, 2019-11-23:
I've not tested PC on simplex recently.

Either way, the TG  identification code does have problems, which is mostly because in the Tier2 version I set the HR-C6000 to receive on both TS1 and T2 simultaneously.
And it has proved to be difficult to distinguish whether the TG reported by the HR-6000 on an incoming signal relates to a TS1 or TS2 signal.

Its because the C6000 only updates its TG output when there is a new signal, and otherwise continues to report the last TG it received, regardless of which TS its receiving at the time.

So I have to check if the TG in the LinConnnect data is different and if so, read the current TS

This works reasonably well, but is not perfect

I don't know how the official firmware does it.

rogerclarkmelbourne, 2019-12-04:
Colins PR fixes this, but I have not pushed a "Latest" sgl that includes this as yet

rogerclarkmelbourne, 2019-12-05:
Fixed

closed 2019-12-05 by rogerclarkmelbourne