rogerclarkmelbourne/OpenGD77 #126: Channel mode displays TS2 when channel has TS1

ajorg, 2019-11-23:
I tried creating two channels per DMR repeater, one on TS1 and the other on TS2. When browsing through channels I see that the display says `DMR TS2` even when the channel is set to TS1. I can verify that the channel appears to be set to TS1 by pressing *Function + Green* to see the Timeslot. The corresponding TS2 channel also shows `DMR TS2`.

If I toggle the Timeslot from the channel details, the display still says `DMR TS2`. If I toggle the timeslot with `*` the display changes to `DMR TS1` (and back when toggled again). But then when I go to another channel, it keeps showing the one that was set with `*`.

It seems like there's a timeslot setting and an *override* timeslot setting, and the display only shows the override setting. I'm not sure which one is actually being used.

Firmware from 6b10f56

alexernst, 2019-11-23:
Take a look at the contact details:

'Contact List' -> select the TG you are using -> Long press GREEN
Is the entry 'Timeslot:' saying 'none' ?

I had the same problem with a codeplug where the contact data was set to a specific timeslot.

rogerclarkmelbourne, 2019-11-23:
The Channels own TS is the lowest priotity when the firmware decides which TS to use

Order of Priority is .. (from highest to lowest)

1. Manual override of TS
2. Contact TG with TS override set
3. Channels own TS

However changing channels should be removing the manual override. 

Do you have a contact TS override defined ?

ajorg, 2019-11-23:
> Is the entry 'Timeslot:' saying 'none' ?

I have only one Contact in my codeplug at the moment. It does say "Timeslot:none".

>  Do you have a contact TS override defined ?

Doesn't look like it.

rogerclarkmelbourne, 2019-11-24:
You're right.

There seems to be a bug in this now

I'll work on a fix

rogerclarkmelbourne, 2019-11-24:
It should be fixed in 69d7ab91022e8f91b6c3870cbd82375d7627b43d

I think its being caused because the current index into the current Rx Group is higher than the number of Contacts in the Rx Group, hence the requested Contact is 0, which does not exist (the numbering starts from 1).

So I've put in 2 different fixes.
If  contact number is invalid it creates a dummy contact for TG9
If the Rx Group index > number of Contacts in the Rx Group list, it will use the primary TG assigned to the channel

rogerclarkmelbourne, 2019-11-24:
The OpenGD77_latest includes this fix

ajorg, 2019-11-24:
This does appear to me to be fixed. Thank you!

rogerclarkmelbourne, 2019-11-24:
OK

closed 2019-11-24 by rogerclarkmelbourne