rogerclarkmelbourne/OpenGD77 #140: Store  and display TG and time in Last Heard

rogerclarkmelbourne, 2019-11-29:


alexernst, 2019-11-29:
This is on my list. 
Also want to add a function to create a new contact from last heard.

rogerclarkmelbourne, 2019-11-30:
TG was already being stored.

I added the time ( from fw_millis()  )  but I've not had time to implement the selection and display control on Last Heard.

For the UI, I think the selected ID / callsign would need to initially be at the top of the screen, but when the down arrow is pressed and eventually the bottom of the list is reached, either the last item in the list would need to be at the top of the screen with 2 blank items under it, or the selected item would need to be the middle and finally the bottom item

It may be easier to implement a system to add 2 blank entries after the end of the list (not actually part of the list)

f1rmb, 2019-12-27:
Hi Roger,

Is "everything" on screen at the same time is a critical feature, or a horizontal scrolling is also good ?.

Cheers.

f1rmb, 2019-12-27:
Just an idea, but the list could use the whole 3 8x16 lines per LH entry (whatever font size used), but don't roll at all.

rogerclarkmelbourne, 2019-12-27:
You mean for LH on the Tx screen ?

I think perhaps having more than just the last callsign is important when in a group QSO.

f1rmb, 2019-12-27:
No, in the Last Heard menu.

rogerclarkmelbourne, 2020-01-13:
The time is now being stored, however we have not worked out the best way to display it

closed 2020-01-13 by rogerclarkmelbourne

rogerclarkmelbourne, 2020-10-04:
This has now been implemented