rogerclarkmelbourne/OpenGD77 #141: Scanning - forced stop leaves user on a channel that was 'skip'

footloosejava, 2019-11-29:
When the scanning is terminated, the user is left on the current channel being looked at.
The channel might also be one that was 'Skip' indicated.

FYI - I ran a test and with 422 channels, all marked 'skip', it looped all channels 20 times in 82 seconds, for a speed of 103 'skip' channels per second.

IW2FHM, 2019-11-29:
Tray to unceck the " Auto Scan " on cps  for all channels

rogerclarkmelbourne, 2019-11-29:
It’s the way Colin wrote the scan function that skip does not really skip, it just scans for a much shorter time.
Colin did this because the firmware crashed if the skip time was too short

I have changed the firmware since Colin wrote the scan function and it’s now less likely to crash, however to cope with having most of the channels marked as skip, would require a bit of a rewrite of the scanning code, as the time taken to load each of hundreds of channels into from Flash and check if the skip flag is set could potentially cause the watchdog timer to timeout and reboot the radio.

Colin’s method to change channel is not ideal, as it directly uses the Up button code, but at least it is working for what most people us the scan for,

Please feel free to submit a PR with a fix for this bug

footloosejava, 2019-11-30:
Ok.Thanks. I see the code now. It loads the channel, then comes back the next call and moves on if 'skip'. Not sure how much time it takes to load, or how to deal with the stop on a 'skip'. That'll take some coffee and a better understanding of everything.

rogerclarkmelbourne, 2019-11-30:
Really, it should not simply be calling the same code that handles the Up

I did change the code so that the function to move Up one channel was handled by a function, but its still the same code which is called when the Up button is pressed.

The RTOS is preemptive multitasking, so in theory you may be able to just keep calling the function which gets the channel data from the codeplug (EEPROM or Flash) but the EEPROM shares the same bus as the RF chip, and the RSSI sampling happens once per OS tick (around every 1mS)

I put in a load of extra code, after Colin wrote the scan feature, to try to prevent bus contention on the I2C bus, and on the whole this now works and does not crash, but I'm not sure whether all the functions pass the return code all the way up the callstack so that you can tell at the top level whether a read from the EEPROM has failed.

Because at the very lowest level of my crash prevention code there is a volatile variable which is used to prevent re-entrance into the I2C code that reads or writes the EEPROM.

I had to use the volatile variable method as a failsafe, because the RF chip is also accessed from the DMR subsystem ISR's and ISR's can't call the RTOS task swap prevention functions.

footloosejava, 2019-11-30:
I meant to ask, what is the best way to get this up and running in Visual Studio? Is there an ideal setup? I've never done C in that and I usually use Intelij Idea for Java.

.. Also, how do you debug and/or run tests on code outside the phone (LOL I just typed phone but I meant Handy Talky!)

How do you step through for debugging?

ea3ihi, 2020-01-15:
Maybe a different scanning strategy is needed, for example, using two variables to hold the current channel and the "next" channel:

While the current channel is being scanned, the channels memory can be read and if it is not a skip channel and not in the nuisance list, save it as "next" channel.

While the scan time elapses, if the "next" next channel is ready, set it as current channel and start looking for the next channel. If the "next" channel is not ready then scan the current channel for a little longer, a couple of ticks more won't hurt.

This way the scan will never stop in a skip channel.

Should the access to the eeprom be guarded using a semaphore, or eeprom access code be inside critical sections?

Does it make sense? If it does i can give it a try

rogerclarkmelbourne, 2020-01-15:
Probably the current method where the scan just uses the Up button functionality to change channel during scanning needs to be replaced with a separate function.

There is also an existing problem with the skipped channels which also needs to be resolved

ea3ihi, 2020-01-16:
Is there a description of the problems with the skipped channles?

ea3ihi, 2020-01-17:
@footloosejava, I have done some changes to the scan functions, please check the version available at:

http://www.opengd77.com/viewtopic.php?f=13&t=501&sid=576f35141846d2120ac9b77d7ecec3ad

David

ea3ihi, 2020-01-25:
I think this issue can be closed after merging the new scan code

rogerclarkmelbourne, 2020-01-25:
OK.

Thanks

closed 2020-01-25 by rogerclarkmelbourne