rogerclarkmelbourne/OpenGD77 #159: In All channel zone key UP is only working with english language setting

alexernst, 2019-12-02:


rogerclarkmelbourne, 2019-12-02:
OK
I will fix

rogerclarkmelbourne, 2019-12-02:
Can you pull and test.

alexernst, 2019-12-02:
That works.

rogerclarkmelbourne, 2019-12-02:
Thanks

I tested in english and french and both seemed to work

BTW. Is there a new way to handle the buttons, or is it just the keys ??

alexernst, 2019-12-02:
At the moment it's only for keys. But it would be easy to include SK1 and ORANGE into the same functions. That would enable at least short and long detection. I think repeat must be filtered.

PTT has to be handled completely  different and SK2 is also special.

rogerclarkmelbourne, 2019-12-02:
OK.

BTW. 
I added a hidden feature of SK_1 + SK_2 pressed at the same time saves the current settings, as when developing it is useful to be able to save the settings before uploading via the debugger,

alexernst, 2019-12-02:
Yes, I already realized that the settings are not saved, since I work with the JLink. The power off in fw_main is never running. 
Perhaps we should  add something like a Debug menu for some 'special' things like saving settings or enable hotspot mode.

rogerclarkmelbourne, 2019-12-02:
Jlink just halts the process then causes some sort of reset.

We could add something to another menu, but pressing SK_1 and SK_2 works to save the settings if I remember I need to save them.

closed 2019-12-02 by rogerclarkmelbourne