rogerclarkmelbourne/OpenGD77 #191: Keypad problems

rogerclarkmelbourne, 2019-12-07:
Just to keep a record of the keypad problems...

My fathers brand new GD-77 which arrived last Wednesday, initially worked OK when I installed the firmware, but the next morning the keypad was not working.

When it wasn't working, one version of the problem was that the each key was beeping on both the Press and also the release.

I think also the keys seemed to have mixed functions as if the keys data was wrong.

But sometimes the keypad didn't work at all.

After the radio has been turned on for a while, the keypad started to work normally.

The problem seems to occur when the radio is turned on in the morning. 
It was not on charge during the night, but it was in the charger all day and would be at 100% battery voltage when taken off charge at about 21:00 local time.
At 06:00 the next morning the keypad didn't work, but the radio was left on, and when it was checked again at 07:00 the keypad was working.

My father has GD-77s, all of them use the OpenGD77 but only one of the radios has this keypad problem.

My father tried leaving the faulty GD-77 in the garage for a few hours, but the radio still worked when it was quite cold e.g. 5 deg C, and it does not get that cold in the bedroom overnight.

So the problem is probably not related to temperature.

rogerclarkmelbourne, 2019-12-07:
As a test, I have installed a version of the firmware which has a 1000 cycle delay between strobing the column select and reading the row.

I also changed the "slew rate" of the Column from Fast to Slow, in case the problem is being caused by "ringing" of the wires in the ribbon cable etc.

But I won't know if this has made any difference until tomorrow morning

rogerclarkmelbourne, 2019-12-07:
Also.

Looking at the key matrix reading code.

In my radio I also tried this

```
uint32_t fw_read_keyboard(void)
{
	uint32_t result = 0;
	int col;

	for (col=3; col>=0; col--)
	{
		GPIO_PinInit(GPIOC, col, &pin_config_output);
		GPIO_PinWrite(GPIOC, col, 1);
	}

	for (col=3; col>=0; col--)
	{
		GPIO_PinWrite(GPIOC, col, 0);
		for(volatile int i = 0; i < 100;i++);// small delay to allow voltages to settle. The delay value of 100 is arbitrary.
		result=(result<<5) | fw_read_keyboard_col();
		GPIO_PinWrite(GPIOC, col, 1);

	}
	for (col=3; col>=0; col--)
	{
		GPIO_PinInit(GPIOC, col, &pin_config_input);
	}

    return result;
}
```
And the key scan still works OK.

i.e This would not reply on pullups, because it drives all the columns high at the begining and then drives each column low, takes the reading and then drives it high again.

Before finally turning off the drive.

Actually since there is no load on the output, I'm not sure whether its necessary to set the columns to input after the read is complete. But it may decrease the current consumption a little.

I have not tried this technique in my father's radio as I wanted to try Slew Rate and a longer delay first

f1rmb, 2019-12-08:
What about your father's radio today ?

rogerclarkmelbourne, 2019-12-08:
I don't know yet.  I tried calling on Skype but no reply

rogerclarkmelbourne, 2019-12-08:
I just got though.

My changes to slew rate and longer delay did not fix it.

Unfortunately its now working again. So I'm unable to test any more fixes.

I will try the code I posted which drives all lines to 1 before strobeing each line to zero, and he will retest tomorrow.

rogerclarkmelbourne, 2019-12-08:
OK My dads radio still didn't work, with teh changes to the slew rate and also to add a longer setting time.

Unfortunately he left the radio on for 10 minutes and then reported it worked OK.

I've now deployed the code I posted previously, and he will tell me tomorrow morning whether the problem is still present.

However he reported his battery level was only around 70% so, I don't understand what factors could cause the keypad to fail if the radio has not been used for an extended period.

Also as a test he removed the battery for 10 mins, refitted it and the radio worked perfectly  

arrrgggh

I can't see any logic to this bug.

f1rmb, 2019-12-08:
Well, that's definitely annoying...

I really need to take a look closer on that code, maybe a stupid idea can emerge ;-)

alexernst, 2019-12-08:
In my repo branch 'keypad-debug' I added a debug screen. The debug screen is available by pressing the Black side key and leaving with Blue. If the debug screen is displayed using the keypad should alter the values. 
If your father's radio is locked a video while using the keypad would be nice. But at least the values of 'state:' and 'scanc:' would be interesting.

There is a option in menuOptions to disable the debug screen. I have set that option to default enable for now.

Additional I reverted fw_keyboard_read() and fw_read_keyboard_col() to the original version.

rogerclarkmelbourne, 2019-12-08:
Thanks. I will try that tomorrow if my last change fails to fix it

rogerclarkmelbourne, 2019-12-09:
Alex

Daniel's changes to the events system cause your keypad-debug to not merge.

I also forgot I have a meeting of the radio club committee to attend, tonight, so probably will not be able to test my fathers radio today.

He will tell me if the forced driving of the strobe lines to High has helped, but as nothing else related to the pullup's has worked I now doubt if that is the problem.

alexernst, 2019-12-09:
I will merge master in my branch and fix the events. 
And if that doesn't help to find the bug, I will build another version with the old key handling.

rogerclarkmelbourne, 2019-12-09:
OK

Thanks

alexernst, 2019-12-09:
Updated my branch with latest master.

rogerclarkmelbourne, 2019-12-09:
Thanks

I tried to contact my father yesterday, but he was out.

But he sent me an email later, saying that the fix did not work.

So I will pull your keyboard debug branch into a branch in my local repo and upload that to his radio tomorrow  (your timezone) (today in my timezone)

rogerclarkmelbourne, 2019-12-10:
I installed the keypad-debug on my fathers radio.

It was not possible to take a video because the LED backlight is set to full and we could not turn it off, as the keypad does not respond.

So I asked my father to tell me whats on the display, and this is what he saw

State 5
k 0:0: 0  b: 0
scanc   21084
debc    21
debs 6318C

No keys were responding so the values did not change when he tried some of the keys

alexernst, 2019-12-10:
OK, I have not much time today, but will think about it.

Could you ask your father to enter the debug screen, when the keypad is working?

Just to check, that the debug screen is working.

rogerclarkmelbourne, 2019-12-10:
OK

I'll tell him to leave it turned on, as that normally fixes the problem.

alexernst, 2019-12-10:
I added another commit to my branch 'keypad-debug'.
Would be nice if you could test that with your father's radio.

21084 means that row 2 is always read as pressed, 6318c is row 2 and row 3. So now pull-up resistors.
So I inspected the code, and, stupid me, I think it's better to enable pull-up resistors on the input lines and not on the output lines. 

Now I'm wondering why anybody could report it's working after enabling the pull-up resistors, as they were in reality still disabled.

rogerclarkmelbourne, 2019-12-11:
@alexernst 

Yes. It should be the input pins.

I didnt noticed that it was the output, as they are just called Row and Column, rather than RowInput and ColumnOutput ...

I'll try adding

```
	PORT_SetPinConfig(Port_Key_Row0, Pin_Key_Row0, &config);
	PORT_SetPinConfig(Port_Key_Row1, Pin_Key_Row1, &config);
	PORT_SetPinConfig(Port_Key_Row2, Pin_Key_Row2, &config);
	PORT_SetPinConfig(Port_Key_Row3, Pin_Key_Row3, &config);
	PORT_SetPinConfig(Port_Key_Row4, Pin_Key_Row4, &config);
```

To the code and see if it fixes my fathers radio.

alexernst, 2019-12-11:
That's what I did in my branch. And renaming them to incclude input and output is a good idea.

rogerclarkmelbourne, 2019-12-11:
Hi Alex

OK

I'll look at you code, I didn't realise you had already updated it.

Also, I will rename them.

rogerclarkmelbourne, 2019-12-11:
Alex. I loaded a version with the correct input pullups onto my fathers radio and it started to work.

So I think this has fixed it.

I've posted a test version to the topic on the forum where some other people are still experiencing the problem.  So hopefully they can test as well.

My father will use the radio today and test again tomorrow morning to confirm this has actually fixed the problem, rather than it just being a random event.

rogerclarkmelbourne, 2019-12-11:
PS. It was too difficult for me to remotely reload the older version which didn't work, but the radio was not working just prior to the upload but worked immediately after the new firmware was loaded.

alexernst, 2019-12-11:
OK I think (hope) this is fixed now. We will see...

Should we leave the debug screen and just set the default to disable in fw_settings.c? 
Maybe it is usable for other not only keypad related bugs later.

rogerclarkmelbourne, 2019-12-12:
My father reported that the radio worked OK this morning when he turned it on, so I think this problem has been fixed.

I've not merged in the debug screen, but I don't think its necessary at the moment to leave it in.

If we get this problem again, I'll build a special version which has the debug screen

closed 2019-12-12 by rogerclarkmelbourne

alexernst, 2019-12-12:
You don't have the pull-ups fixed in master?

rogerclarkmelbourne, 2019-12-12:
Good point.

I will meed to merge by local branch which just used your change to the keypad init function.

Things have been very busy this week and my PC had major problems after a Windows update, so I forgot to merge that fix into the master