rogerclarkmelbourne/OpenGD77 #213: Blue button changes to TG of incoming signal even if keypad is locked

rogerclarkmelbourne, 2019-12-10:


jangelor, 2019-12-10:
I kinda think of this as a feature. Maybe implement it differently if the functionality is caused by a bug?

f1rmb, 2019-12-11:
I will check that, maybe it's a side effect of the new event thing.

Cheers

f1rmb, 2019-12-11:
Fix is in PR #220 

Cheers.

testbird, 2019-12-11:
I would also see good use in a mode that still allows the operation with the current radio settings, i.e. locking only the keypad plus any channel switching side buttons, but leave PTT plus any talkgroup adjustments, manual scan stop/resume functions etc. on the side buttons working.

* unlocked
* keyboard (settings) locked (some side changes button filtered/disabled)
* PTT-only
* everything locked (possibly even volume if possible)

f1rmb, 2019-12-11:
Hi,

   You can't lock the keypad while scanning (ATM), as any input, with few exceptions, just stops the scanning.

Cheers.

testbird, 2019-12-11:
Cheers! Yes, I would see that current behavior as improvable. I'm just not 100% sure if you meant your comment as an agreement or dismissal.

testbird, 2019-12-11:
BTW: Could be cool to also even let the blue button (function) ***plus*** any shortcut key still work while the keyboard (or better channel? settings) can stay locked.

testbird, 2019-12-11:
A related Question about the new keypad code, does it now allow for this:

Somewhere, I don't remember where, I read about using a more "phone standard" way for using the green (call) and red (cancel) buttons.

And the idea to allow short presses of the blue button (function or shortcut) to start (with a beep+screen indication) a two second timeout for the user to press the next function defining/finalizing button.

testbird, 2019-12-11:
Please, still respond to my posts above, but I just had an idea for fixing this issue and the function key conflict (not usable for other things during RX).

What if the action to switch to the currently heard TG would be a short press on the PTT during RX? [At least if polite TX restrictions are enabled? Or always function+PTT? (during the 2 sec timeout as above?]
* would only be locked together with PTT
* function (+ other keys) would always work the same

closed 2019-12-15 by rogerclarkmelbourne