rogerclarkmelbourne/OpenGD77 #224: Report that VHF Rx performance is worse after the Calibration fix

rogerclarkmelbourne, 2019-12-12:
Riku reports that the VHF Rx performance is worse after the calibration data offset fix was applied.

This possibly implies that there is something wrong with the way that the calibration data from Flash is being intrepreted and applied to the hardware.

Its not entirely clear in fw_calibration, the exact function of the settings as Kai's names for them does not seem to correspond to Jason's names for the same memory locations in the Flash memory.

I think we need to establish precisely what these settings control and whether they affect Rx or Tx or both.

testbird, 2019-12-12:
A reason for annoying intermittent AF interruptions in a SDR based radio, I heard with another radio model, were said to occur due to changes in the signal level that caused the device to re-scale its sensitivity or dynamic range, which was causing an interruption in the AF.

So suboptimal range calibration settings, together with usual random signal or noise changes, and hickups in the re-scaling code may be one possibility.

riku2015, 2019-12-13:
i was having this issue hotspot in one meter away. no signal changes. also on repeater with external antenna (was different, see videos on forum). it seems changing frequency on radio will help on breakups. more talk with videos about this at forums: https://www.opengd77.com/viewtopic.php?f=11&t=151

testbird, 2019-12-13:
Ok, good findings. That close in the near-field though, do you hear squeak sounds in analog FM?

riku2015, 2019-12-13:
i think no, squeak means the re-scale sensitivity..? it comes on factory firmware but not on opengd77, i need test more on car while moving, but it is way better on analog than factory one especiality with ctcss rx.

testbird, 2019-12-13:
>i think no, squeak means the re-scale sensitivity..?

No sure, these SDR other radios have just short RX interruptions even if they are just moved around in shouting distance.
With the squeak I was looking for another hypothetical explanation. I actually do not know how those squeak tones come about  in the near-field. Does somebody here know? Just thought they could possibly interfere if on the same frequency as the DMR keying, and slightly shifting the carrier freq would split them appart.

In any case in the near-field I think the (possibly only almost) standing wave is a different thing than a RF wave passing by. So triggering by changes in the field strength may not be completely ruled out, yet.

closed 2019-12-15 by rogerclarkmelbourne