rogerclarkmelbourne/OpenGD77 #243: Enabling optimisation on fw_HR-C6000.c may degrade DMR audio

rogerclarkmelbourne, 2019-12-15:
See.  

People seem fairly sure that the unoptimised DMR code is better.

https://www.opengd77.com/viewtopic.php?f=13&t=333

rogerclarkmelbourne, 2019-12-15:
There is potentially a problem with when the timeslot number (known as the TimeCode or TC in the C6000 data sheet) , is read from register 0x52

Reg 0x52 contains 1 bit that contains the TC ( 0 = TS 1, 1 = TS 2)

The colour code is also in register 0x52

```
	rxColorCode 	= (tmp_val_0x52 >> 4) & 0x0f;
	timeCode =  ((tmp_val_0x52 & 0x04) >> 2);
```

Register 0x52 is read at the start of the System interrupt and also in the Data Received function, which is called from the System Interrupt if the sub-type (of interrupt status) is Data Received 

Register 0x82 contains the sub-type of interrupt when then the System Interrupt signal is generated by the C6000

However, it seems that sometimes the value of register 0x52 is different when its read in those 2 different locations, which would cause problems.

The datasheet describes reading 0x52 to check the Colour Code

`2.	when receiving sys_inter, read reg0x51 and reg0x52, if reg0x51[7:4] is equal to local cc  `

However I can't see anything in the datasheet that describes when the TimeCode should be read

rogerclarkmelbourne, 2019-12-16:
There is definitely something wrong with the current code.

If I remove the code that reads 0x52 in the Data Received function, and just use the value from the top of the System Interrupt function, the timecode seems to be reversed during Tx.

If I reverse the TC when I read it (i.e TC = 1- TC) then the Tx is on the correct TS but Rx audio does not work.

I will need to investigate this in more detail

testbird, 2019-12-17:
> something wrong with the current code

It may also be the hardware treating RX and TX inversely. It could, e.g. allow to re-transmit the data without any changes, just as it is (i.e. no checksum updates, if there is something like that.)

rogerclarkmelbourne, 2020-01-18:
The changes Colin made to the TimeCode / TS handling seem to have fixed this problem and fw_hr-c6000.c is now running optimised and reports are that the problems no longer exist

closed 2020-01-18 by rogerclarkmelbourne