rogerclarkmelbourne/OpenGD77 #272: Tx screen is not removed when PTT is released if Tx to repeater fails wake up the repeater

rogerclarkmelbourne, 2019-12-19:


alexernst, 2019-12-20:
I will take a look at this.

rogerclarkmelbourne, 2019-12-20:
Thanks

Its a bit intermittent.

I noticed this when I was accidentally on the wrong frequency this morning.

But when I test it now, the problem has vanished.

alexernst, 2019-12-20:
In commit #c1c67de he added some changes in lines 134ff in uiTxScreen.c.
Now if <code>trxIsTransmitting == false</code> handleEvent() is only called every 200 ticks. But in DMR mode handleEvent() checks for <code>slot_state<DMR_STATE_TX_START_1</code>. And only then the screen is removed.

But I don't know what Daniel wants to achieve here.
So this one goes to Daniel.

f1rmb, 2019-12-20:
Hi,

   I wanted to limit jumping into handleEvent() (as everywhere else in menu/ui system).
So I've added condition checking for that, but it seems I missed failure ones.

Cheers.

rogerclarkmelbourne, 2019-12-20:
OK.

That would make the screen removal dependent on timing between when the PTT was released and whether the DMR was sending the Wake-up pulse or waiting for the response to the Wake-up

f1rmb, 2019-12-20:
Yup.

   In the PR I've raised, handleEvent won't wait for that if PTT is released.

Cheers.

rogerclarkmelbourne, 2019-12-21:
Fixed

closed 2019-12-21 by rogerclarkmelbourne