rogerclarkmelbourne/OpenGD77 #284: Support for GD-77 clones

rogerclarkmelbourne, 2019-12-21:
Just to document this.

The Baofeng DM-1801 seems to be a non-exact clone of the GD-77, and currently the OpenGD77 firmware won't run correctly on the D-1801.

Physically the case of the radio looks similar, but has some obvious differences including separate VFO and A/B buttons as well as re-organised arrow keys.

The battery looks identical to the GD-77 but they are not quite interchangeable.
The DM-1801 will only fit onto the GD-77 if pushed very hard. So there must be some small differences in the physical shape of the battery.

Neither Tx or Rx works using the OpenGD77 firmware.

It looks like the At1846 is being configured correctly, because if you tune to a frequency the RSSI meter reads correctly. But no audio can be heard.

It sounds as if the audio amplifier is being enabled, but the audio routing switch is probably not routing the audio output to the amplifier

So it looks like a lot of the control lines from the MCU to the other parts of the radio are different.

If anyone has a schematic for the DM-1801 or wants to 'buzz' those connections, it may be possible to work out the DM-1801 and get the OpenGD77  firmware working on this hardware

jumbo77, 2020-02-28:
i gonna to buy a DM-1801 open it for connecton testing .

rogerclarkmelbourne, 2020-02-28:
@jumbo77 

OK.

You need a small TORX screwdriver to open the case

I bought a cheap set from ebay and it works fine

"25 IN 1 Small MINI REPAIR PRECISION SCREWDRIVER TORX TOOL KIT SET FIX"

e.g. 
https://www.ebay.com.au/itm/25-in-1-Precision-Screwdriver-Set-Tool-Kit-Mini-Small-Torx-PC-Laptop-Repair-Fix/333337955847

jumbo77, 2020-03-01:
@rogerclarkmelbourne i got the DM-1801 now ,i open it ,and take photo ,the Audio PA is TDA2822M,it power on by AO3401 MOS FET,so i think now power control on TDA2822M ,i try to find CONTROL point now.

jumbo77, 2020-03-01:
with MD380 UHF also use TDA2822M ,i check the schema ,i find it CONTROL by STM32F405 pin96 AFCO,but i don't find MK22FN512 the same point ,i double check now.

jumbo77, 2020-03-01:
i found it , MK22FN512 pin 78 PTC6/LLWU_P10 is AFCO signal,OK!

Jumbo/BG7IYN

jumbo77, 2020-03-01:
@rogerclarkmelbourne  i double check the Direction of MK22FN512,small dot is  pin 1,100 corner ,not two big dot.
correct  pin  is **53 PTB0/LLWU_P5** now .

jumbo77, 2020-03-01:
pin 27 DAC0_OUT/TCMP1_IN3/ADC0_SE23 is APC/TV power control ,the same point on MD380 UHF.

jumbo77, 2020-03-01:
i think DM1801(pcb is BF1801-RF-V2.5 2018-10-16) is the simple version of MD380,so we can reference MD380 's hardware ,i think hardware is  no problem on it.hope DM1801 will working on opengd77 firmware soon,i'm waiting for,i have MD760 working OpenGD77 very well,than you!

rogerclarkmelbourne, 2020-03-01:
@jumbo77 

If your DM1801 working ?
What changes did you makes ?

For me, the FM Rx audio signal is very low, and beeps and DMR don't work at all.

I changed the power control to the TDA2822M onto its pin 2 is working OK. on PTB0 (same as the GD-77).

Are you sure the amp control is on PTC6, I'm sure this is the amp input select, (DMR or FM), and I have tested this connection and it seems to be correct.

jumbo77, 2020-03-01:
@rogerclarkmelbourne 
i didn't tested OpenGD77 fw yet,i don't know which cps for dm1801,and no sgl for it yet.

rogerclarkmelbourne, 2020-03-02:
Ah.

When you load the OpenGD77 DM-1801 firmware, you can use the OpenGD77 CPS which includes the tools for calibration etc etc

I can post a SGL, here but its easy to build for DM-1801, as there is a now a "Configuation" in the OpenGD77 MCUExpresso project to build for DM-1801 including generating the SGL file etc

rogerclarkmelbourne, 2020-03-02:
[OpenDM1801_202003021124.zip](https://github.com/rogerclarkmelbourne/OpenGD77/files/4272829/OpenDM1801_202003021124.zip)

Here is the latest build from my local repo.

jumbo77, 2020-03-02:
i don't change any,just open it and check the circuit,only hardware check.  PTC6 maybe is amp input select, (DMR or FM) ,i didn't check that ,i checked  pin 27 DAC0_OUT/TCMP1_IN3/ADC0_SE23 is APC/TV.

rogerclarkmelbourne, 2020-03-02:
Yes.

PTC6 is the selection for source to the amplifier, I tested this and its exactly the same as the GD77.

So far the only differences I found were the red and green LEDs, but I have not checked the C6000 connections.

Please email me roger at rogerclark.net and I will send you some information I can't post online

Also. Look in the GD-77 source

https://github.com/rogerclarkmelbourne/OpenGD77/search?q=PLATFORM+%3D%3D+DM-1801&unscoped_q=PLATFORM+%3D%3D+DM-1801

And you'll see the pin definitions for both the GD77 and DM-1801.

I think the RF Rx control is correct, as the radio can receive the signal, the EEPROM is working,  but I don't know if the Flash chip is working.

Actually I can test the Flash chip, because the OpenGD77 CPS can read the flash chip,
I will do this soon.

If the Flash chip is not working, it would explain a lot of problems because the Calibration is in the Flash !

rogerclarkmelbourne, 2020-03-02:
PS. Where are you. and Do you have a Ham callsign ?
Mine is VK3KYY (which I'm sure you know)

jumbo77, 2020-03-02:
i'm in china ,QTH is ShenZhen,my callsign is BG7IYN,i like mmdvm,and have MD380x2,XTS5000x1,FT-2D,LimeSDR,Hackrf,MD760,RTLSDR,nanoVNA,UHSDR,all ham things.

rogerclarkmelbourne, 2020-03-02:
@jumbo77 

Do you think there is any chance Baofeng would give you the connections ?

I think they would sell more radios if the OpenGD77 was on the 1801

I also have a RD-5R which is similar hardware, but a different display

jumbo77, 2020-03-02:
i register account on opengd77.com ,i will check the W25Q16 SPI flash working,i think they would,because more radio sell that opengd77 on it,currently OpenGD77 on MD760/GD77 it work better then MD-380,because you can input freq on it,and you can select 250mw-5w power,but 380 not,so 380 is no  popular later ,i think,i will sell my old MD-380 and change to MD-760 and DM 1801 ,haha.

rogerclarkmelbourne, 2020-03-02:
OK. I see your account.

I will create a section for DM-1801

BTW. It would be possible to port to MD-380 but its more complex becuase the MD-380 is STM32F4 MCU not MK22, so it would require a separate project (not using MCUExpresso)

rogerclarkmelbourne, 2020-03-02:
I sent you a PM

rogerclarkmelbourne, 2020-03-02:
I just checked and Flash is not working

Probably its not connected to the same pins.

This would explain why a lot of functions do not work correctly.

The calibration will be completely wrong !

I will try to trace the connections, it should be easy, as there are only 3 connections

rogerclarkmelbourne, 2020-03-02:
@jumbo77 

I tried to trace the connections, and I think the values are

#define Port_SPI_FLASH_CS_U  	PORTE
#define GPIO_SPI_FLASH_CS_U  	GPIOE
#define Pin_SPI_FLASH_CS_U   	6

#define Port_SPI_FLASH_CLK_U 	PORTE
#define GPIO_SPI_FLASH_CLK_U 	GPIOE
#define Pin_SPI_FLASH_CLK_U  	5

#define Port_SPI_FLASH_DI_U  	PORTE
#define GPIO_SPI_FLASH_DI_U  	GPIOE
#define Pin_SPI_FLASH_DI_U   	4

#define Port_SPI_FLASH_DO_U  	PORTA
#define GPIO_SPI_FLASH_DO_U  	GPIOA
#define Pin_SPI_FLASH_DO_U   	19

But this does not work :-(

Using the GD-77 values, the response from the Flash was always zero, but now the response is always 0xFF

So perhaps I made a mistake with the connections

@jumbo77  can you check your PCB

rogerclarkmelbourne, 2020-03-02:
@jumbo77 

Please pull from the OpenGD77 master. It has updates for the core code (GD-77).

I will look again at the Flash memory problem, and attach a logic analyser to confirm the signals are correct.

rogerclarkmelbourne, 2020-03-02:
@jumbo77 

I have found the problem with the flash pinout.

I had transposed the DO and DI pins

I've updated the repo.

I used the OpenGD77CPS to read the lower 1Mb of Flash, (I will need to modify the CPS to read all 2Mb)

And the Calibration table is not in the same address as in the GD-77, it seems to be at 0x70000 instead of 0x8F000

So I have added conditional compilation for the calibration table base address.

I did a quick hack on the OpenGD77CPS to read the calibration table from 0x70000 and it seems to have the same format as the GD-77 but I have not compared in detail with my GD-77 radios.

FM Rx seems slightly better now, but is far from perfect.  I may try installing the calibration table from one of my GD-77 into the DM-1801 at 0x8f000, because I don't think 0x8f000 is used in the DM-1801 firmware.

**NOTE.

Do not upload the DMR ID using the CPS, because I think it may overwrite the calibration table in the DM-1801**

rogerclarkmelbourne, 2020-03-02:
Actually, It looks like the address is 0x6F000 and 0x70000 is a backup.

The format of the data is also possibly slightly different from the GD-77 which would be annoying if it is

The master oscillator value in my calibration data is very low, and sets the radio around 25kHz off frequency.

jumbo77, 2020-03-02:
i trace the connections today ,
it is : 
define Port_SPI_FLASH_CS_U PORTE
#define GPIO_SPI_FLASH_CS_U GPIOE
#define Pin_SPI_FLASH_CS_U 6

#define Port_SPI_FLASH_CLK_U PORTE
#define GPIO_SPI_FLASH_CLK_U GPIOE
#define Pin_SPI_FLASH_CLK_U 5

#define Port_SPI_FLASH_DI_U PORTE
#define GPIO_SPI_FLASH_DI_U GPIOE
#define Pin_SPI_FLASH_DI_U 4

#define Port_SPI_FLASH_DO_U PORTA
#define GPIO_SPI_FLASH_DO_U GPIOA
#define Pin_SPI_FLASH_DO_U 19

i check it was correct!

also know:
HC_6000 to MK22FN 512 Trace is 
HC_6000 pin 34  DI     to   MK22FN512 pin A13
HC_6000 pin35  DO    to    MK22FN512 pin A15
HC_6000 pin 36 CLK   to   MK22FN512 pin A14
HC_6000 pin 37  CS    to  MK22FN512  pin A16
-----------------------------------------------------------
24C512   and RDA1846s SCK  to  MK22FN512  PE24
24C512   and RDA1846s SDA to  MK22FN512   PE25

rogerclarkmelbourne, 2020-03-02:
RDA1846 and EEPROM are already working , because they are on the same pins.

Thanks for the connections to some of the C6000, but there are many more connections between the MK22 and the C6000

I will try to find them today.

rogerclarkmelbourne, 2020-03-02:
BTW. To fix the master oscillator offset problem, perhaps we should just add an offset to the value read from the calibration. Perhaps the offset is the same for all radios.

I don’t know why the firmware requires this offset. I will look at the oscillator calibration code to see if there is any special calculation used by the GD-77

rogerclarkmelbourne, 2020-03-03:
@jumbo77 

I checked those I2S connections between the C6000 and the MK22 and they are the same as the GD-77

jumbo77, 2020-03-03:
UHF Tx works,that is some GPIO controls works,don't worry,i will try to take a bare pcb photo,to Reserve it,so we can find trace more easy,i will upload to OpenGD77.

rogerclarkmelbourne, 2020-03-03:
Yes.

I think the GPIO controls for the PA must be different

I found that the 4 GPIO's which control the VHF and UHF Tx PA and the VHF and UHF Rx amplifier are on pins

PTE0
PTE1
PTC13
PTC15

But I don't know which pin matches to which function

On the GD-77 PTE0 is the DMR reset pin, which I have not changed on the DM-1801 so there are 2 things using the same pin  :-(

I will need to find were the DMR SLEEP and DMR RESET pins are connected on the MK22 on the DM-1801

jumbo77, 2020-03-03:
I found 
DMR PWD_DOWN is  E3

rogerclarkmelbourne, 2020-03-03:
excellent

I will look for RESET unless you found it already

rogerclarkmelbourne, 2020-03-03:
I think C6000 RESET is E2

rogerclarkmelbourne, 2020-03-03:
Success

My radio now has startup beeps !!

rogerclarkmelbourne, 2020-03-03:
I've pushed an update for this.

jumbo77, 2020-03-03:
OK!

rogerclarkmelbourne, 2020-03-03:
BTW. DMR Rx does not work.

So I will need to check the interrupt signals from the C6000 to the MK22

51 RF_RX__INTER
50 RF_TX__INTER
49 SYS_INTER
48 TIME_SLOT_INTER

also probably need to check
58 U_CS
But this may be the same as on the GD-77

rogerclarkmelbourne, 2020-03-03:
Also. There is now less current being taken, 112mA on Rx with the display at 100% which is similar to the GD-77 (which uses 100mA)

rogerclarkmelbourne, 2020-03-03:
Ah,

Also vocoder interface will need to be found.

But I will look for C6000 interrupt pins first

jumbo77, 2020-03-03:
i try to find :
51 RF_RX__INTER
50 RF_TX__INTER
49 SYS_INTER
48 TIME_SLOT_INTER
yesterday ,but can't find the MK22FN512 connection,seem no connect ,but it must connect it.

rogerclarkmelbourne, 2020-03-03:
I use a multimeter continuity tester

rogerclarkmelbourne, 2020-03-03:
I will look for those pins now

rogerclarkmelbourne, 2020-03-03:
I think the SYS and TS interrupts are the same pins,  but possibly they are reversed on the DM-1801

I think those 2 signals are on C17 and C18, but it is very hard to measure this as the pins are very small.

I think I need to solder a wire to each pin, so that its possible to confirm this.

rogerclarkmelbourne, 2020-03-03:
@jumbo77 

I had a wrong setting in my options

DMR Rx is actually working.

The pins for the interrupts are probably the same as in the GD-77.

I have not tested Tx yet, but DMR Rx is definitely working

rogerclarkmelbourne, 2020-03-03:
I think I may have now fixed the DMR Tx !

jumbo77, 2020-03-04:
ok,that's good news!

rogerclarkmelbourne, 2020-03-04:
I have pushed an update.

DMR Tx is now working 

BUT.,  I have no power on UHF.
I don't know if I damaged the UHF PA when I was testing, or whether the GPIO for the UHF PA is not correct.

VHF output is working OK, but I don't know if I am enabling the VHF PA or the UHF PA.
But definitely one of the PA is working

jumbo77, 2020-03-04:
ok,I will try to upload fw to 1801 and test the TX,RX,UHF,VHF,TX Current,OUTPUT power try to Test it ,to ensure to problem.

jumbo77, 2020-03-04:
i bought 2 x1801 for test ,it arrived today.

rogerclarkmelbourne, 2020-03-04:
I checked both VHF and UHF Tx,

On both VHF and UHF FM, I set the radio to 500mW and it consumes around 600mA on Tx, but on UHF there is no power output to the antenna socket. :-(

So I would be interested to know if you get any power on UHF.

I think perhaps my UHF output FET is blown and is a short circuit :-(

rogerclarkmelbourne, 2020-03-04:
I retested the current consumption.

On 50mA  VHF only consumes 100mA, but on UHF the radio is consuming 800mA

I think one of the PA FETS has probably blown and is a short circuit :-(

I will try to find which device is blown, but it will be a long time before I can make a repair, because I can not buy replacements from a local company.

I will probably need to order a replacement from AliExpress.

If the DM-1801 is similar to the GD-77 

This final FET as a  AFT05MS004NT1  There is also a driver FET a 2SK3078    before that is a bipolar NPN   2SC4226 

I think probably the final is blown.

rogerclarkmelbourne, 2020-03-04:
umm.

I checked all the FETs but none is short circuit.

This very strange.

jumbo77, 2020-03-04:
jumbo@jumbo-ThinkPad-T410:~/DM1801$ sudo python gd-77_firmware_loader.py --firmware=OpenDM1801_202003021124.sgl --model=DM-1801
[sudo] password for jumbo:
Now flashing your DM-1801 with "OpenDM1801_202003021124.sgl"
Firmware file confirmed as SGL
 - Sending Download command
 - Sending ACK
 - Sending encryption key
 - Sending F-PROG command
 - Sending radio modem number
 - Sending radio modem number 2
 - Sending version
 - Sending erase command
 - Send post erase command
 - Sending Program command
 - Sending last block
Firmware update complete. Please reboot the DM-1801.

ok,first time to upload successed!

jumbo77, 2020-03-04:
i will test it use constant-current power ,limit current to TX(1A),it won't short cut to broken. also 1801 have fast recover fuse may be 3A.

rogerclarkmelbourne, 2020-03-04:
My linear PSU will only supply 1A. 

I checked all the power FET's and they are all the same resistance.

The current consumed on UHF is not dependent on the DAC power level control. 
Even the 50mW power level, which is very low DAC output voltage, consumes 800mA on UHF (but only 100mA on VHF).

So I think perhaps the short is on the UHF PA master power supply rail.

jumbo77, 2020-03-04:
1801-BF fw       7.5V supply             1801-GD77(OpenDM1801_202003021124.sgl)
0.08A                  RX_V                       0.16A 
0.03-0.06A          RX_V_Save              0.16A
0.97A                  TX_L_VHF                don't test
2.4A                    TX_H_VHF              don't test
0.86A                  TX_L_UHF               don't test

rogerclarkmelbourne, 2020-03-04:
You should compile from the latest commit in the repo

I made a change today to fix C6000 interrupt pins to the MK22

Select the DM 1801 configuration before you build the project

jumbo77, 2020-03-04:
update new lastest commit in repo now. testing.

jumbo77, 2020-03-04:
updated latest fw,DMR RX LocalRepeater Hear Voice ,it seem no different to 760,tx test it.

[42 more replies]

closed 2020-04-02 by rogerclarkmelbourne