rogerclarkmelbourne/OpenGD77 #314: Audio "Flutter" when Mobile

na7q, 2019-12-31:
When you are mobile and there is a lot of signal fluctuation, the receiving stations audio flutters at a certain "chopping" rate if you will.. The signal strength is strong enough to decode without issue.  
When comparing one radio with OpenGD77 and stock firmware, the stock does not present the same issues at all. Telling me it's related to the firmware. Both present on VHF and UHF repeaters.

Our area is very hilly and mountainous.  The signal fluctuation is from a minimum of S5 to maximum of S9 according to the RSSI and S meter.

riku2015, 2019-12-31:
i think this is related to this closed issue https://github.com/rogerclarkmelbourne/OpenGD77/issues/224 I have no idea why that was closed, as it is not resolved, people still have problems.
Here is also more info related to this https://www.opengd77.com/viewtopic.php?f=11&t=151
I can agree with you, when mobile the flutter is alot worse still. changing frequency by few Hz it helps abit, but comparing to stock firmware it is way worser. Isues is VHF specific, but i can agree on uhf it has also problems.

na7q, 2019-12-31:
Well I hope it will be looked at again. VHF is at times incredibly rough. UHF is not as bad, but not great either.

na7q, 2020-01-04:
I was fighting this REALLY hard this evening while mobile. UHF or VHF, it was pretty horrific. I was even using an external 1/4 wave on 2M style antenna. While my stock firmware GD-77 had no problems at all internally in the vehicle with no external antenna. It's as if a squelch is just way too high in DMR mode. The light flickers rapidly with fluctuating signals. Way worse than stock ever did.

rogerclarkmelbourne, 2020-01-04:
There is no squelch on DMR in the traditional sense.

On New Years Day, I opened one of my GD-77's and have soldered some 0.25m thick wires onto the data bus between the MCU and the C6000 chip, to which I have connected a USB logic analyser, and there appears to be be some configuration of the C6000 which is missing from the OpenGD77 firmware.

Multiple blocks of 32 bytes are sent to the C6000 ever time the channel is changed.

```

.000000000000000,0x84,0x06,0x21
0.000029200000000,0x04,0x06,0x23
0.000064880000000,0x84,0x88,0x00
0.000094240000000,0x03,0x00,0x00,0x00,0x04,0xCE,0x08,0x92,0x0A,0x7B,0x0A,0x20,0x07,0x94,0x03,0x64,0xFE,0x79,0xF9,0xE2,0xF6,0x9D,0xF5,0x60,0xF6,0x70,0xF9,0x91,0xFE,0x16,0x03,0x05,0x07,0x4D
0.000310880000000,0x84,0x88,0x00
0.000340240000000,0x03,0x00,0x0A,0x00,0x0A,0x8A,0x08,0xCC,0x05,0x26,0x00,0x63,0xFB,0x8B,0xF7,0xAA,0xF5,0x96,0xF5,0xC2,0xF8,0x26,0xFC,0x3C,0x01,0x23,0x05,0xCB,0x09,0x31,0x0A,0x9B,0x09,0xB9
0.000556880000000,0x84,0x88,0x00
0.000586280000000,0x03,0x00,0x06,0xBD,0x02,0x4B,0xFD,0x5A,0xF8,0xFC,0xF6,0x22,0xF5,0x6A,0xF6,0xFD,0xFA,0x82,0xFF,0x37,0x04,0x18,0x08,0x16,0x0A,0x54,0x0A,0x55,0x08,0x1B,0x04,0x20,0xFF,0x40
0.000802920000000,0x84,0x88,0x00
0.000832280000000,0x03,0x00,0xFA,0x89,0xF7,0x01,0xF5,0x6B,0xF6,0x1F,0xF8,0xF6,0xFD,0x53,0x02,0x43,0x06,0xB6,0x09,0xB6,0x0A,0x9C,0x09,0x36,0x05,0xD2,0x01,0x2B,0xFC,0x44,0xF8,0x2C,0xF5,0xC4
0.001063640000000,0x84,0x88,0x01
0.004951720000000,0x84,0x88,0x00
0.004981080000000,0x03,0x00,0xF5,0x94,0xF7,0xA5,0xFB,0x84,0x00,0x5B,0x05,0x1F,0x08,0xC7,0x0A,0x89,0x0A,0x03,0x07,0x53,0x03,0x0D,0xFE,0x1E,0xF9,0x97,0xF6,0x73,0xF5,0x60,0xF6,0x99,0xF9,0xDB
0.005203600000000,0x84,0x88,0x00
0.005233000000000,0x03,0x00,0xFE,0x70,0x03,0x5C,0x07,0x8E,0x0A,0x1E,0x0A,0x7D,0x08,0x97,0x04,0xD5,0x00,0x08,0xFB,0x39,0xF7,0x72,0xF5,0x85,0xF5,0xDC,0xF8,0x65,0xFC,0x93,0x01,0x7E,0x06,0x17
0.005449680000000,0x84,0x88,0x01
0.009951760000000,0x84,0x88,0x00
0.009981120000000,0x03,0x00,0x09,0x5E,0x0A,0x9F,0x09,0x93,0x06,0x75,0x01,0xF1,0xFD,0x02,0xF8,0xB8,0xF6,0x02,0xF5,0x74,0xF7,0x2F,0xFA,0xD1,0xFF,0x93,0x04,0x6C,0x08,0x50,0x0A,0x68,0x0A,0x3F
0.010203640000000,0x84,0x88,0x00
0.010233040000000,0x03,0x00,0x07,0xDF,0x03,0xCA,0xFE,0xE4,0xFA,0x3B,0xF6,0xD2,0xF5,0x64,0xF6,0x42,0xF9,0x3C,0xFD,0xAC,0x02,0x9D,0x06,0xFC,0x09,0xDA,0x0A,0x95,0x09,0x06,0x05,0x84,0x00,0xD0
0.010449680000000,0x84,0x88,0x01
0.014950360000000,0x84,0x88,0x00
0.014979720000000,0x03,0x00,0xFB,0xEF,0xF7,0xEF,0xF5,0xAD,0xF5,0xA8,0xF7,0xDF,0xFB,0xD8,0x00,0xB7,0x05,0x6E,0x08,0xF9,0x0A,0x93,0x09,0xE3,0x07,0x0F,0x02,0xB5,0xFD,0xC4,0xF9,0x4F,0xF6,0x4D
0.015198480000000,0x84,0x88,0x00
0.015227880000000,0x03,0x00,0xF5,0x63,0xF6,0xC5,0xFA,0x26,0xFE,0xCB,0x03,0xB3,0x07,0xCE,0x0A,0x38,0x0A,0x6C,0x08,0x5F,0x04,0x83,0xFF,0xAC,0xFA,0xE7,0xF7,0x3D,0xF5,0x77,0xF5,0xF9,0xF8,0xA6
0.015444560000000,0x84,0x88,0x01
0.019948920000000,0x84,0x88,0x00
0.019978320000000,0x03,0x00,0xFC,0xEA,0x01,0xD9,0x06,0x61,0x09,0x88,0x0A,0x9F,0x09,0x6A,0x06,0x2B,0x01,0x97,0xFC,0xAB,0xF8,0x76,0xF5,0xE4,0xF5,0x81,0xF7,0x63,0xFB,0x22,0xFF,0xEF,0x04,0xBF
0.020198480000000,0x84,0x88,0x00
0.020227880000000,0x03,0x00,0x08,0x88,0x0A,0x79,0x0A,0x25,0x07,0xA0,0x03,0x74,0xFE,0x89,0xF9,0xEF,0xF6,0xA5,0xF5,0x60,0xF6,0x68,0xF9,0x83,0xFE,0x05,0x02,0xF5,0x07,0x41,0x09,0xFA,0x0A,0x8C
0.020444520000000,0x84,0x88,0x01
0.024949040000000,0x84,0x88,0x00
0.024978400000000,0x03,0x00,0x08,0xD5,0x05,0x35,0x00,0x74,0xFB,0x9B,0xF7,0xB4,0xF5,0x99,0xF5,0xBE,0xF8,0x1B,0xFC,0x2D,0x01,0x12,0x05,0xBD,0x09,0x29,0x0A,0x9A,0x09,0xC0,0x06,0xCA,0x02,0x5B
0.025198560000000,0x84,0x88,0x00
0.025227920000000,0x03,0x00,0xFD,0x6B,0xF9,0x09,0xF6,0x28,0xF5,0x69,0xF6,0xF4,0xFA,0x74,0xFF,0x27,0x04,0x08,0x08,0x0B,0x0A,0x50,0x0A,0x59,0x08,0x26,0x04,0x2F,0xFF,0x51,0xFA,0x98,0xF7,0x0A
0.025444600000000,0x84,0x88,0x00
0.025473960000000,0x03,0x00,0xF5,0x6D,0xF6,0x19,0xF8,0xEA,0xFD,0x42,0x02,0x33,0x06,0xA9,0x09,0xAF,0x0A,0x9C,0x09,0x3E,0x05,0xE0,0x01,0x3C,0xFC,0x54,0xF8,0x37,0xF5,0xC9,0xF5,0x91,0xF7,0x9A
0.025690600000000,0x84,0x88,0x00
0.025719960000000,0x03,0x00,0xFB,0x75,0x00,0x4B,0x05,0x10,0x08,0xBD,0x0A,0x86,0x0A,0x08,0x07,0x5F,0x03,0x1D,0xFE,0x2F,0xF9,0xA5,0xF6,0x7B,0xF5,0x60,0xF6,0x91,0xF9,0xCD,0xFE,0x60,0x03,0x4D
0.025942720000000,0x84,0x88,0x01
0.029977840000000,0x84,0x88,0x00
0.030011200000000,0x03,0x00,0x07,0x83,0x0A,0x18,0x0A,0x7F,0x08,0xA1,0x04,0xE4,0x00,0x18,0xFB,0x48,0xF7,0x7C,0xF5,0x88,0xF5,0xD7,0xF8,0x59,0xFC,0x83,0x01,0x6D,0x06,0x09,0x09,0x56,0x0A,0x9E
0.030227840000000,0x84,0x88,0x00
0.030257200000000,0x03,0x00,0x09,0x9A,0x06,0x82,0x02,0x02,0xFD,0x12,0xF8,0xC4,0xF6,0x08,0xF5,0x72,0xF7,0x26,0xFA,0xC3,0xFF,0x82,0x04,0x5D,0x08,0x46,0x0A,0x64,0x0A,0x43,0x07,0xEA,0x03,0xDA
0.030473880000000,0x84,0x88,0x01
0.034949040000000,0x84,0x88,0x00
0.034978400000000,0x03,0x00,0xFE,0xF5,0xFA,0x4A,0xF6,0xDA,0xF5,0x65,0xF6,0x3C,0xF9,0x2F,0xFD,0x9B,0x02,0x8C,0x06,0xF0,0x09,0xD3,0x0A,0x97,0x09,0x0F,0x05,0x92,0x00,0xE1,0xFB,0xFE,0xF7,0xFA
0.035198600000000,0x84,0x88,0x00
0.035227960000000,0x03,0x00,0xF5,0xB1,0xF5,0xA4,0xF7,0xD4,0xFB,0xC8,0x00,0xA6,0x05,0x60,0x08,0xF0,0x0A,0x91,0x09,0xE9,0x07,0x1C,0x02,0xC5,0xFD,0xD4,0xF9,0x5C,0xF6,0x54,0xF5,0x62,0xF6,0xBD
0.035444640000000,0x84,0x88,0x01
0.039949080000000,0x84,0x88,0x00
0.039978440000000,0x03,0x00,0xFA,0x18,0xFE,0xBB,0x03,0xA4,0x07,0xC2,0x0A,0x34,0x0A,0x70,0x08,0x6A,0x04,0x92,0xFF,0xBD,0xFA,0xF6,0xF7,0x46,0xF5,0x7A,0xF5,0xF4,0xF8,0x9A,0xFC,0xDA,0x01,0xC8
0.040198640000000,0x84,0x88,0x00
[... cut ...]
0.075233480000000,0x03,0x00,0xF8,0xDD,0xF6,0x13,0xF5,0x6F,0xF7,0x13,0xFA,0xA6,0xFF,0x61,0x04,0x3E,0x08,0x31,0x0A,0x5D,0x0A,0x4B,0x08,0x00,0x03,0xF9,0xFF,0x16,0xFA,0x66,0xF6,0xEB,0xF5,0x68

```

We are working blind with the C6000 chip, as there is very limited documentation on this chip, which is produced by a very secretive Chinese company, which is part of one of the universities in China, and is probably government funded.

However, hopefully analysing the data between the MCU and the C6000 will eventually allow us to operate the C6000 more effectively, even though we don't know what the data or commands actually do.

na7q, 2020-01-04:
The unfortunate reality. I imagine with time, things will be figured out. Maybe someone will toss some black magic on the subject and make it happen. If not, that's okay too. I understand how these things can be incredibly complex. 
Thanks for doing all that you do to try to figure it out and make it work.

riku2015, 2020-01-04:
na7q, i can confirm the "squelch" thing having signal over 50% and it is breaking up with green light.  i have 2x gd77 it happens on both.

Roger, nice job :) it is like network analyzer, but logic level :D :D i hope something will turn out, there is something missing for sure, maybe the rx-amp drive?

footloosejava, 2020-01-04:
Def a lot of picket fencing on analog when driving. 
May be related.

riku2015, 2020-01-16:
this might be a possible fix to the sluttering problem. please upgrade to the latest and test it :)
https://github.com/rogerclarkmelbourne/OpenGD77/commit/4dd6a789af0d01431372c9812556932c658730a6#diff-74b5b5e9570efc5c0553bb327cd41940

rogerclarkmelbourne, 2020-01-18:
I think this has probably been fixed in the versions that now have a 1 frame DMR audio buffer

closed 2020-01-18 by rogerclarkmelbourne