rogerclarkmelbourne/OpenGD77 #332: Band limits are only correct for some regions

ajorg, 2020-01-06:
Ham radio band limits vary by region. 430-450 in Australia, 420-450 in US (430-450 North of "Line A"), 430-440 in the UK. 144-148 in ITU regions 2 and 3, but 144-146 in region 1.

OpenGD77 currently limits TX to either the physical limits of the radio or a [hard-coded 430-450 and 144-148](https://github.com/rogerclarkmelbourne/OpenGD77/blob/d777de283647035121369ea48e58c334555f0913/firmware/source/functions/fw_settings.c#L28) depending on your "Band limit" setting, which means that unless you disable band limits you can't transmit between 420 and 430 if allowed (as in most of the US), and there's no way to prevent transmitting between 146 and 148 if you're operating in ITU region 1 (Europe, Africa, Russia, etc.).

The stock firmware uses a codeplug setting to define some band limits. If the OpenGD77 CPS defaulted to 430-450 and 144-146 as globally *safe* defaults when creating a new codeplug, and the firmware used the configured bands as the band limits, operators in other regions would be better served.

rogerclarkmelbourne, 2020-01-09:
I agree the band limits may as well be the US limits and also include the 220Mhz band, as the VHF limits are already too wide for most users in Europe.

Trying to re-use the codeplug band limits is problematic, as people could have already messed around with those values when using the official firmware, so we can't simply check whether those values are the defaults and then ignore them and use the US band limits, and otherwise use them as the band limits

The only safe way to do this is to store band limits in the new data section in the bottom of Flash, which I've started to implement for the boot melody and boot image.
But it would take a lot of work to make the whole thing bulletproof

rogerclarkmelbourne, 2020-04-30:
I can't see a way to fix this, because using the codeplug data could cause problems

closed 2020-04-30 by rogerclarkmelbourne