rogerclarkmelbourne/OpenGD77 #337: Desktop GIT Error

DL7PJ, 2020-01-08:
[image]

RobertMTalbot, 2020-01-08:
That was my fault- I committed a file with the backslash in the name. I should have recognized that it would cause issues for Windows users. Sorry about that.

It should be fixed now as Roger removed that file.

DL7PJ, 2020-01-08:
Error is still there, will wait, thanks for the quick reply

DL7PJ, 2020-01-08:
I removed and reloaded OpenGD77 Git. Error is still there.
[image]

[image]

RobertMTalbot, 2020-01-08:
The master branch in Roger's repo shows that this has been removed.

[image]

DL7PJ, 2020-01-08:
[image]

DL7PJ, 2020-01-08:
[image]

RobertMTalbot, 2020-01-08:
Yes, again my apologies for that. It was my fault. Roger sent me a message about it this morning (US time) and said he removed that file and the he got it working fine again. I suppose we'll have a wait a few hours until he chimes in.

DL7PJ, 2020-01-08:
Thank you for the quick answer, no problem is a hobby and not a high availability system, we will just wait and see 73 de Peter

rogerclarkmelbourne, 2020-01-08:
I fixed it last night, but you may need to delete your repo and clone it again.

I had to delete my local copy, because it was broken by that file.

DL7PJ, 2020-01-08:
hi Roger I tried several times and had no success. I deleted all data several times and reloaded it from the git. The error keeps coming back, see screenshots

DL7PJ, 2020-01-08:
https://github.com/rogerclarkmelbourne/OpenGD77/issues/337#issuecomment-572148188

rogerclarkmelbourne, 2020-01-08:
This seems to be a bug in the GitHub Desktop application.

I'm not seeing the same bug in the GitHub command line application, and I have successfully cloned the repo again.

Perhaps the GitHub desktop application is retaining some sort of cache.

I don't think there is a way to retrospectively remove one file from inside a git repo once its been added.

I've now removed the file from whats visible in the current version, and thats the version which the git client should try to update in your local file system.

rogerclarkmelbourne, 2020-01-09:
See

https://github.com/git-for-windows/git/issues/2435

rogerclarkmelbourne, 2020-01-09:
Also see 

https://github.com/zephyrproject-rtos/zephyr/issues/21426

DL7PJ, 2020-01-09:
Hi Roger , thank you, now I can update. I close the issue 73 de Peter 

[image]
[image]

closed 2020-01-09 by DL7PJ

rogerclarkmelbourne, 2020-01-09:
OK.

Did a new version of the App fix it ?

DL7PJ, 2020-01-09:
The error now only appears when I click on an entry in the history, 
but update is no longer a problem
73 de Peter
[image]

[image]

[image]