rogerclarkmelbourne/OpenGD77 #343: Request:  Minimum backlight setting

khaytsus, 2020-01-10:
I would love to see the ability to set the minimum backlight setting, such as 0% (off; as it is today), 10%, 20%, etc, so you can always see the screen but dimmed when it's idle.

ea3ihi, 2020-01-10:
This requires touching some parts of the code

- Set the backlight value to a previously saved value instead of 0, this is the easy one.
- Add the new setting for the bakligh off value
- Add a new item to the settings menu so one can set the value in a similar way as the standard "on" mode backlight setting

At first, taking into account that a lot of work will consist in copying what is already available for the displayBacklightPercentage settings it does not look difficlult to implement and can be a good way to start learning how things work in the firmware....I will give it a try, i am preparing the development environment as I type.

ea3ihi, 2020-01-10:
Pull request #346 created

#346

ea3ihi, 2020-01-15:
@khaytsus I have posted a test version with this feature in the forum:

http://www.opengd77.com/viewtopic.php?f=13&t=492&sid=a0aaff228c70e5abeb1777e1d0f1f550

rogerclarkmelbourne, 2020-01-18:
This is now part of the master code

closed 2020-01-18 by rogerclarkmelbourne