rogerclarkmelbourne/OpenGD77 #380: Random reboots

G0FHM, 2020-01-17:
Latest release (16/01) sees the radio reboot at random, and display settings return to default, with backlight timer reverting to "always on".

rogerclarkmelbourne, 2020-01-18:
There was a bug in the backlight cancel function, but the other reason that the settings return to the value prior to the reboot is that the settings are only saved when the radio is turned off.

If its reboots, it won't save the settings first.

There is a feature used by the developers to force the settings to be saved. 
Press the Black and Blue buttons at the same time, and the current settings will be saved.

I'm also going to release another version today, if its stable, which includes more optimisations.
This may or may not be more stable than the version on 16th.

rogerclarkmelbourne, 2020-01-18:
Can you try the code optimised version I just posted to the forum

https://www.opengd77.com/viewtopic.php?f=13&t=505

closed 2020-01-20 by rogerclarkmelbourne