rogerclarkmelbourne/OpenGD77 #427: Setting the Contact option to "None" in a DMR channel does not work

jacopo-j, 2020-02-08:
Hello. The issue occurs when I create a DMR channel (using the community CPS) and leave the "Contact" option to "None":

[image]

According to the help text,

> Selecting the None option prevents a call from being initiated on the channel.

However, what actually happens on the radio is that the first contact from the Contacts list is associated to the channel:

[image]

and nothing prevents me to initiate a call.

f1rmb, 2020-02-08:
I would check "Rx Only" instead.

Cheers.

jacopo-j, 2020-02-08:
@f1rmb That does prevent me from making a call. However, my contact #1 is still shown on the cannel's screen, and I feel like it's not supposed to. Is this the intended behaviour?

rogerclarkmelbourne, 2020-02-08:
This is the intended behaviour 

The digital contract field is not used unless the Rx Group list is set to none.

The Rx Group list data is now used as a list of talk groups associated with the channel, it’s no longer used as the Rx Group list.

Did you read the documentation ?

rogerclarkmelbourne, 2020-02-08:
Edit.

You can’t set both the contact and Rx group list to none.

It’s not valid for a digital channel not to have a TG.

jacopo-j, 2020-02-08:
Okay then, I will close this. Thank you.

closed 2020-02-08 by jacopo-j