rogerclarkmelbourne/OpenGD77 #431: Virus/malware warnings from downloaded files

skaufman, 2020-02-10:
Virus/malware detection apps for both Winders and MacOS report viruses/malware in the files from this project. What is going on?

f1rmb, 2020-02-10:
Hi,
  You're not the only to experience such false positive.
Don't worry, if you download the firmware from the github repository, there is no problem at all, hundreds of people are using the same file.

Cheers.
...
Daniel

skaufman, 2020-02-10:
Thanks for the follow-up. However a friend who downloaded the files a couple weeks ago encountered no warnings but now he does get them. Not so sure you can assume that these are false positives. 

Sent from my iPhone, so plz xcuse any ttpyos or abbrvns.

f1rmb, 2020-02-10:
That's normal, it's not the same file, as you can get new version each few days.

rogerclarkmelbourne, 2020-02-10:
The antivirus companies update their software every few days.

I have similar issues on multiple projects, and also get the same problem for my day job.

See any example of a file that had not changed for 3 years and had been download by thousands of people, started to get reported as a virus

https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/716

Take a look at some of the various posts on the false positives e.g.

https://weblog.west-wind.com/posts/2016/Oct/05/Dealing-with-AntiVirus-False-Positives

And you'll see what a broken state, the world of antivirus companies, is now in.

PS. Even signing an exe, it still gets reported as a virus, because the exe's I create as part of my day job get signed and distributed by one of my clients. And even with the file being signed with an extensive digital certificate, they can still get miss identified  as a virus.

rogerclarkmelbourne, 2020-02-10:
I decided to leave this open, in case anyone else has the same problem, as its not possible to fix this.

IW2FHM, 2020-04-15:
Hi,
By dowload the file "OpenGD77CPSInstaller.exe" from R20200414 on my computer (W7, antivirus AVG) is segnalling virus " Win32:Adware-gen[Adw]. It is true??

rogerclarkmelbourne, 2020-04-30:
Cant fix this, as its outside of our control

closed 2020-04-30 by rogerclarkmelbourne