rogerclarkmelbourne/OpenGD77 #450: reducing RX selection errors with disabled CC filter

testbird, 2020-02-26:
* To avoid DMR decoding the pure RF noise into disturbing audio, may it make sense to set the squelch to a non-zero value to blank the noise when CC filtering is turned off?

* It seems to me the histogram data to deal with CC fluctuations may better be cleared with each RF signal (squelch?) drop?
I mean only resetting the histogram data when the frequency changes would work for frequencies used by an unknown repeater, but maybe not so good for simplex frequencies that are shared by groups using different CCs in an intermittent fashion. What if group 1 makes a lot of transmissions over time and group 2 only a few? Will the occasional group 2 transmissions with a different CC be properly decoded?

testbird, 2020-02-27:
Possibly a better idea for a dynamic CC selection:

The assumption is that the noise produces random distributed CC values over time.

Then maybe it could also work to just shorten the histogram interval to say 360ms and have the "listen" level of say 50% frames with the same CC, to start the audio output. This then might allow to trigger quickly enough to thoroughly catch any real received signal.

rogerclarkmelbourne, 2020-02-29:
Closing as all reports from users have been OK

closed 2020-02-29 by rogerclarkmelbourne

testbird, 2020-03-02:
Ok, as known problem, reports are foreseeable, though.

For example, when operators are on mountains where they can receive different repeaters on the same frequency, or very low disturbing signals, or unhearable neighbor channel stuff enough to get decoded, ...

> I could possibly preload the histogram with a value to bias it towards the CC that is defined for the channel, but this would slow down the detection of other CC's and also in your case I doubt it would fix the problem.

I would still think maybe using just a few hundred ms "signal locking histogram" and resetting it everytime with the squelch (RX signal end), but saving the selected CC together with the "last heard" for TX might work more general.

testbird, 2020-03-02:
And using the channel's CC if there is no "last heard CC", of course. Maybe a CC "hang time".

testbird, 2020-03-02:
Hm, might even work simpler, by resetting the histogram right after the short signal locking interval, no need to wait for RX ending.

testbird, 2020-03-03:
Thinking of a better way to make the CC selection histogram work also for changing signals, but even without requiring a signal squelch, made me think of this:

What if the histogram would continuously cover only the last 12 or 13 (odd) received frames, simply dropping the oldest CC code as a new one comes in?
=> Then only turn on the audio when some CC reaches a threshold.

TX defaults to channel's CC + the configured last heard hang time (includes 0 and -1).

testbird, 2020-03-07:
Is there a reason for the non-replying to ideas and issue closing?
The new post on the forum made me wonder.