rogerclarkmelbourne/OpenGD77 #451: `git rev-parse --short HEAD` is run hundreds of times each build

ajorg, 2020-02-27:
I've been trying to figure out how to fix this on my own but MCUXpresso (Eclipse) is evading me. I know why the command is run (so that the `GITVERSION` will show up on the Firmware details) but it shouldn't need to run once for each compiled file. There has to be a way to configure the build so that this value is either pulled directly from IDE (which knows about Git and which revision the working directory is currently on) or pulled via commandline once and then used throughout the build without running the command again.

rogerclarkmelbourne, 2020-02-27:
I agree this is not ideal

You could use a prebuild step to put that value into a file and then during the build you'd need to echo the value in the file. 

But that would probably take roughly the same amount of processing 

I'm not sure how MCUExpressor builds the makefile, but I saw this 

https://stackoverflow.com/questions/44038428/include-git-commit-hash-and-or-branch-name-in-c-c-source

Which had a way of making the git tag into a variable in the makefile

BTW.

I'm not sure if I'm just imagining things, but since I added the second build configuration, every time I press the Debug button to upload and run the firmware via a SWD debugger, it seems recompile everything, even if I've already just compiled.

The Build process works OK, its just the Debug stuff that has started to do this

Fortunately, I use a 4GHz Core i7 machine, which can do a full build in a few seconds, but otherwise it would be a complete pain.

rogerclarkmelbourne, 2020-04-02:
No practical solution

closed 2020-04-02 by rogerclarkmelbourne